robotology/mesh-support #1985: Right hip roll motor smoked

isorrentino, 2024-12-04:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

While using the robot the right hip roll motor smoked, but we were not asking for high motor currrents. So, we do not have any log or strange error, only the overcurrent from the motorgui.

### Detailed context

Is it possible to double check if this motor is still working fine or it needs to be replaced?

Thanks a lot.

cc @Gandoo @Fabrizio69 @fbiggi

Gandoo, 2024-12-05:
ciao @isorrentino @AntonioConsilvio we checked the continuity of the 3 phases of the motor and they are ok, then we tried the movement with the box and it is ok, then reconnected and it seems to work.

cc @filippoborgogni @Fabrizio69 @maggia80

isorrentino, 2024-12-09:
This problem is related to https://github.com/robotology/icub-tech-support/issues/1990.

We have determined that the motor burnouts were not caused by the experiments but rather by the new 2FOC firmware version released in https://github.com/robotology/icub-firmware-build/releases/tag/v1.41.0

cc @Fabrizio69 @DanielePucci @maggia80 @ale-git @Gandoo @AntonioConsilvio @GiulioRomualdi @fbiggi

AntonioConsilvio, 2025-03-19:
The problem was due to a malfunctioning firmware. We checked the phases and they are good, and with the version of firmware mentioned in this [comment](https://github.com/robotology/icub-tech-support/issues/1990#issuecomment-2737029596) the robot can walk without problems.

Closing!