robotology/mesh-support #1986: Issue on right ankle pitch temperature sensor

S-Dafarra, 2024-12-05:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

The joint goes in hardware fault because of clearly bogus readings

### Detailed context

Here the error messages:
```
	792.369441	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=89s 129m 524u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1771)
	797.371887	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=94s 129m 528u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1850)
	802.372689	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=99s 130m 515u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1823)
	807.370921	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=104s 131m 516u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1763)
	812.369560	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=109s 132m 529u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1692)
	817.373152	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=114s 133m 524u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1620)
	822.369941	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=119s 134m 526u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1552)
	827.373559	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=124s 135m 524u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1480)
	832.379394	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=129s 136m 524u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1415)
	837.375143	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=134s 137m 526u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1355)
	842.378303	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=139s 138m 530u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1300)
	847.381157	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=144s 139m 526u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1249)
	852.377282	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=149s 140m 534u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1201)
	857.385540	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=154s 141m 532u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1159)
	862.393090	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=159s 142m 538u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1120)
	867.383103	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=164s 143m 526u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1084)
	872.386231	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=169s 144m 524u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1052)
	877.386147	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=174s 145m 532u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=1021)
	882.381356	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=179s 146m 529u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=994)
	887.094266	ERROR			from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=183s 856m 718u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_ankle_pitch (NIB=0), Raw_temperature_value=970)
```

### Additional context

This joint is using a i2c extender. 

cc @maggia80 @valegagge @MSECode

S-Dafarra, 2024-12-05:
> Hi [[image]@S-Dafarra](https://github.com/S-Dafarra) , I'm looking into that. Just to check the diagnostic, on which version of icub-main is running the robot? Need to see if the high level filtering is correct.

We just updated to ``master``. It is 2.7.1

MSECode, 2024-12-05:
Then, do u have logged some data for the temperature. So we can check how data have been read and how they changed before the errors, to see if we are still reading badly.

S-Dafarra, 2024-12-05:
> Then, do u have logged some data for the temperature. So we can check how data have been read and how they changed before the errors, to see if we are still reading badly.

No, it was immediately after startup

MSECode, 2024-12-05:
> > Then, do u have logged some data for the temperature. So we can check how data have been read and how they changed before the errors, to see if we are still reading badly.
> 
> No, it was immediately after startup

Ok, I would first check the cable, if it might be broken or detached. If that not the case I'll dig on the problem.

valegagge, 2024-12-05:
Hi @MSECode ,
Can you provide the temperature values in degree starting from the ones provided in the log?
Thank heaps

MSECode, 2024-12-05:
If I recall correctly, on ergocub we only have motors with PT100 temperature sensors. Thus, the data shown on the logs are respectively the following temperatures in Celsius:
```
142.511
148.944
146.744
141.859
136.084
130.233
124.712
118.871
113.602
108.743
104.293
100.168
96.2889
92.8965
89.7481
86.8432
84.2622
81.7629
79.5869
77.6533
```
I would think that they are false positives. I'm just worried about 2 thing:
1. We actually do not have any negative (error value) reading
2. The temperatures collected seems to decrease following a certain trend

However, if the robot is still working fine and you started to read those values at a startup of the system, which was previously idle it cannot be a real problem of overtemperature. If, instead, the robot seemed to have problem during the calibration phase, that might be reasonable value but you should not be able to use the motor at its total power now (if it started to burn on the stator you should at least see worse performances on the joint ).

Thus, in the case those are false positive, we need to check the cabling --> I would say that the most likely option is that we have the i2c extender or the tdb itself not well connected and so we see disturbed data.
In the case the joint has some issue in the encoder reading during the calibration, considered that the `ankle_pitch` calibrates at the hard stop, 2 situations might be happened:
- encoder at the joint cannot read well, while encoder at the motor is fine --> during calibration motor moved without the joint going anywhere raising its temperature
- calibration went wrong at we pushed towards the hard stop limit until calibration timeout 

For the latter two cases for sure we have encoder errors in the yri diagnostic, which can help in checking if we had issues on the encoders.

S-Dafarra, 2024-12-05:
Oh shoot, maybe they were not fake at all, see https://github.com/robotology/icub-tech-support/issues/1987. When I saw the values I thought they were already in Celsius.

I thought they were fake cause it was happening right at startup and the joint was calibrating well. I wonder if instead, it was an issue of the board that is in short circuit.

MSECode, 2024-12-05:
> ight at s

For sure I'll update the diagnostic message to give you both values (raw and converted) for clearance.

MSECode, 2024-12-05:
> I thought they were fake cause it was happening right at startup and the joint was calibrating well. I wonder if instead, it was an issue of the board that is in short circuit.

In that case I would expected more errors (the negative values you can see here: https://icub-tech-iit.github.io/documentation/temperature_sensors/software/dataflow/#error-handling). However, if a resistor is damaged the output can be have an offset since you have not a balanced voltage divider anymore. But it is important that the resistor is just damaged. In the case of an open circuit you'll get an error.

valegagge, 2024-12-05:
Hi @S-Dafarra ,
I didn't understand if how this issue is related to what happened in the #1987 . Thanks for the clarification. :)

S-Dafarra, 2024-12-06:
I am not sure either, but in any case I suppose that there is some issue on the right ankle 2foc.
In the meantime I restored the temperature sensor

S-Dafarra, 2024-12-13:
In the end, the motor was indeed burning. So I guess that the temperature sensor was working fine, and I got misled by the fact it was happening with the robot just turned on and by the raw value I assumed was in Celsius

AntonioConsilvio, 2024-12-17:
Hi! As can be seen in this comment, the motor was burnt out:

- https://github.com/robotology/icub-tech-support/issues/1987#issuecomment-2541028575

So, closing this issue due this comment 🚀:

- https://github.com/robotology/icub-tech-support/issues/1987#issuecomment-2548427538

closed 2024-12-17 by AntonioConsilvio