robotology/mesh-support #1988: power button cover lost

jimingda, 2024-12-06:
### Robot Name 🤖

iCubShanghai01 S/N:049

### Request/Failure description

Hi, 

I've lost a power button cover like the one shown in the picture below. Could anyone provide the exact details about it so I can purchase a replacement?

Thanks.

[image]

### Detailed context

None

### Additional context

None

filippoborgogni, 2024-12-06:
Hi @jimingda, 
the power button cover  you are looking for is not sold individually , but it is sold together with the power button.
You can buy it directly from Farnell's web site. 
The product code is 2280381 and the manufactured code is 1FS091. 
I enclose a screenshot of the site for easier understanding. Closing!

[image]

jimingda, 2024-12-09:
Thank you. @filippoborgogni

closed 2024-12-09 by jimingda