robotology/mesh-support #1989: Errors with R1 hands (ergomorphing)

fbrand-new, 2024-12-06:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

I just turned on the robot with the new hands calibrated by @MSECode and I have these errors:
[image]
I do not know if it is safe to proceed further, or if I should turn off the roboto

@AntonioConsilvio @Fabrizio69 @MSECode 

### Detailed context

See above

MSECode, 2024-12-06:
Hi,
can u send me the configuration file of the calibrator and  the one of the motion control, that I'll double check the calibration parameters and the PID values. 
Thanks

fbrand-new, 2024-12-09:
Update with the log files of the yri from inside and outside the docker on the base
[log.txt](https://github.com/user-attachments/files/18061235/log.txt)
[log_no_docker.txt](https://github.com/user-attachments/files/18061236/log_no_docker.txt)

mfussi66, 2024-12-10:
The OP issue regarding i2t limit seems a side-effect of the calibration procedure being unable to read the FAP data. The sensors read just fine and we can see it in the port `/cer/left_hand/POS<2 or 4>:o` if the service is enabled in the yarprobotinterface config file.

maggia80, 2024-12-10:
- embobjPOS was not enabled in CCMake in the r1-base  
- POS services were not enabled in the yarprobotinterface.ini file 

we still have issues in the right hand:

 [image]

MSECode, 2024-12-11:
> In order to run the calibration procedure correctly, as I talked with [@MSECode](https://github.com/MSECode) , the POS service needs to **be active** before the calibrator. If the opposite happens, the operation fails.
> 
> In ergoCubSN002 the POS service starts right at the beginning of the yarprobotinterface file:
> 
> https://github.com/robotology/robots-configuration/blob/master/ergoCubSN002/ergocub.xml#L12-L22 .
> 
> Instead, in R1SN003 it starts later:
> 
> https://github.com/robotology/robots-configuration/blob/master/R1SN003/R1.xml#L36-L46
> 
> and therefore the time between the POS and the calibrator is not sufficient.
> 
> I'm not entirely sure how the yarprobotinterface works, but I'm afraid there is some kind of race condition in this scenario.

Exactly, or at least that is what I've always seen on `ergoCub` and `iCub` robots, with the POS service running first. Thus, that can be said to be the standard use case. I'm not sure, anyways, if that is mandatory for having the whole system running correctly but what @mfussi66  said is totally correct.

randaz81, 2025-01-13:
@mfussi66 @AntonioConsilvio can you provide and update on this activity?

mfussi66, 2025-01-13:
Hi @randaz81 , 
The issue was solved by:

- Enforcing the priority at startup of of the POS service over the calibrators
- Tuning the PID controllers

I'll do a PR in robots-configuration ASAP.

I'm not sure if some mechanical maintainance is required, @AntonioConsilvio might know more about this.

mfussi66, 2025-01-20:
PR here: https://github.com/robotology/robots-configuration/pull/705

The right ring+pinky is not moving correctly, @AntonioConsilvio is on it.

edersonferrante, 2025-01-24:
- [x] We found the broken pinky phalanx and replaced it ( Cod. 17391)

 [image]

https://github.com/user-attachments/assets/5d8a22e4-800d-477a-9db6-b539ce1dbce7
@Fabrizio69

AntonioConsilvio, 2025-01-24:
Here the PR with the new calibration of right pinky:

- https://github.com/robotology/robots-configuration/pull/707

I tested the robot and it works properly. 

Now I am going to close this issue ✅. If other problems appear, feel free to open new issues!

closed 2025-01-24 by AntonioConsilvio