robotology/mesh-support #1990: Joints go very quick into overcurrent

S-Dafarra, 2024-12-06:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

After updating the FW to the [latest version](https://github.com/robotology/icub-firmware-build/blob/v1.41.0/info/firmware.info.xml), we noticed that the joints go suspiciously too fast into overcurrent.

 

### Detailed context

We noticed this while walking. After half a step, both hip rolls went into hardware fault because of overcurrent. Here the log:
[log_yarprobotinterface_hf_hip_roll.txt](https://github.com/user-attachments/files/18038874/log_yarprobotinterface_hf_hip_roll.txt)

Interestingly, we also noticed the same behavior immediately after calibration if the hardware limits are a bit too restrictive with respect the value used for calibration.

We noticed the same also on the SN000

### Additional context

cc @maggia80 @marcoaccame @valegagge @MSECode @isorrentino

S-Dafarra, 2024-12-06:
Downgrading the 2foc version to 3.3.36 seemed to fix the issue

Nicogene, 2024-12-06:
The changes between `3.3.36` and `3.3.37` are the ones contained in this PR:
- https://github.com/robotology/icub-firmware/pull/524

with @ale-git we will check the next week what is the possible cause, we will probably contact you in order to understand what it the problem.

We probably need also a `2024.11.2` once the fix is done (cc @pattacini, @traversaro), sorry for the unconvenience.

Nicogene, 2025-01-09:
Hi @S-Dafarra,

the PRs containing the fix have been merged, and the repos has been tagged:
- https://github.com/robotology/icub-firmware-build/releases/tag/v1.41.1
- https://github.com/robotology/icub-firmware/releases/tag/v1.41.1
- https://github.com/robotology/icub-firmware-build/pull/187
- https://github.com/robotology/icub-firmware/pull/551

@ale-git verified that on the robot, once you double-check it we can close this issue.

We will also do a distro 2024.11.2 containing those new tags (cc @traversaro)

cc @maggia80

S-Dafarra, 2025-01-09:
> [@ale-git](https://github.com/ale-git?rgh-link-date=2025-01-09T09%3A01%3A50.000Z) verified that on the robot, once you double-check it we can close this issue.

Just to understand, in which robot has been tested? Is the firmware already flashed the boards?

ale-git, 2025-01-09:
Hi @S-Dafarra, I've tested it on SN 000 legs. After I've restored the .36 firmware, thus the robots have to be updated.

ale-git, 2025-02-10:
I've checked the 3.3.38 firmware on ergoCub 000 standing, stressing and shaking it back and forth and sideways, without faults.

AntonioConsilvio, 2025-03-19:
> I've checked the 3.3.38 firmware on ergoCub 000 standing on its feet, stressing and shaking it back and forth and sideways, without faults.

Great! Closing 🚀