robotology/mesh-support #1991: Ethernet connector disconnected

gabrielenava, 2024-12-09:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

Same as https://github.com/robotology/icub-tech-support/issues/1972: the connector disconnected again from the board. Would it be possible to fix it?

[image]
[image]

### Detailed context

cc @DanielePucci @Fabrizio69

Fabrizio69, 2024-12-10:
Hi @gabrielenava , can you describe how it happened? This time the connector was soldered and we can't do better. At the moment I don't have a solution to permanently solve this strange problem in a place where there are not even moving parts...

gabrielenava, 2024-12-16:
Concerning 

>Hi @gabrielenava , can you describe how it happened? This time the connector was soldered and we can't do better. At the moment I don't have a solution to permanently solve this strange problem in a place where there are not even moving parts...

I noted that the once mounted, the connector is very close to the metal support of the dummy battery (see the above picture). The support of the network board is in plastic, so, if one pushes too hard when connecting the ethernet and/or the support is pushed during iRonCub experiments, the connector will hit the metal support. This can explain why it broke twice in short time

[image]

anyways, @AntonioConsilvio gave me another network board and I mounted it on the robot. The ethernet connection now works, the problem is solved! we can close the issue

cc @Fabrizio69

AntonioConsilvio, 2024-12-17:
Nice, closing! 🚀

closed 2024-12-17 by AntonioConsilvio