robotology/mesh-support #1993: OccasionalROP error on left arm

S-Dafarra, 2024-12-10:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

During Humanoids, we noticed the following error:
```
[ERROR] embOBJerror(): errtype = TRACE from EOobject =
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2,
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 28 12 656
[ERROR] embOBJerror(): errtype = TRACE from EOobject =
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2,
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 20 596 656
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is 59 and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2,
[ERROR] embOBJerror(): errtype = TRACE from EOobject =
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 4,
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 28 12 656
[ERROR] embOBJerror(): errtype = TRACE from EOobject =
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2,
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 59514 32681 656
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is 48 and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 4,
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [ERROR] embOBJerror(): errtype = TRACE from EOobject = [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 4, [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 28 0 656 [WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is 38 and can overflow [WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is 38 and can overflow [DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 4, [ERROR] embOBJerror(): errtype = TRACE from EOobject = [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 28 12 656 [ERROR] embOBJerror(): errtype = TRACE from EOobject = [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 20 596 656 [WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is 26 and can overflow [DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [ERROR] embOBJerror(): errtype = TRACE from EOobject = [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 28 20 656 [DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [ERROR] embOBJerror(): errtype = TRACE from EOobject = [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 28 12 656 [ERROR] embOBJerror(): errtype = TRACE from EOobject = [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos = 20 596 656 [DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 2, [ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=673s 532m 783u : HW: strain values saturation. 3 is the channel involved. In the last se
```

This is something I don't recall seeing before.

### Detailed context

I see in the error messages that the board 10.0.1.4 have been involved. This board have been substituted in https://github.com/robotology/icub-tech-support/issues/1983

I guess these might be related, so I opened this mainly for documentation.

cc @marcoaccame @valegagge

AntonioConsilvio, 2024-12-18:
Using the robot I got the error, later I found one ETH connector damaged and @AntonioAzocar restored it. After I tested the robot and it was not happened.

However, I'm leaving the issue open a little longer waiting for feedback on it!

SimoneMic, 2025-01-03:
Probably still ongoing see #2006