robotology/mesh-support #1994: Generic motor error on right arm

S-Dafarra, 2024-12-10:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

During Humanoids, and more recently during some tests in the lab, we experienced a ``Generic Motor Error``
```
[INFO] from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=2145s 341m 85u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([ ], [ 1314 ]) 
[INFO] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=2145s 388m 435u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([ ], [ 13 ]) 
[INFO] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2145s 440m 746u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([ ], [ 13 ]) 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2146s 333m 885u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] Provided non-increasing time vector 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2147s 335m 895u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2148s 337m 900u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2149s 339m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2150s 341m 891u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2151s 343m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ).
[WARNING] Provided non-increasing time vector 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2152s 345m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2153s 347m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2153s 581m 79u : MC: generic motor error. (Joint=r_shoulder_pitch (NIB=0) (Errors:CAN_RXIsPasv CAN_IsWarnTX CAN_IsWarnRX CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2153s 581m 256u : MC: generic motor error. (Joint=r_shoulder_roll (NIB=1) (Errors:CAN_RXIsPasv CAN_IsWarnTX CAN_IsWarnRX CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2153s 583m 76u : MC: generic motor error. (Joint=r_shoulder_pitch (NIB=0) (Errors:CAN_IsWarnTX CAN_IsWarnRX CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2153s 583m 254u : MC: generic motor error. (Joint=r_shoulder_roll (NIB=1) (Errors:CAN_IsWarnTX CAN_IsWarnRX CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2153s 592m 76u : MC: generic motor error. (Joint=r_shoulder_pitch (NIB=0) (Errors:CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2153s 594m 118u : MC: generic motor error. (Joint=r_shoulder_roll (NIB=1) (Errors:CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2154s 38m 132u : MC: 2FOC quadrature encoder dirty. The number of thicks in a complete revolution of the motor was lower than expected, the optical disks need to be cleaned. (Joint=r_shoulder_pitch (NIB=0), Raw_quad_encoder_value=0) 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2154s 349m 896u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] Provided non-increasing time vector 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2155s 351m 895u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2156s 353m 891u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2157s 355m 885u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2158s 357m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] Provided non-increasing time vector 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2158s 581m 76u : MC: generic motor error. (Joint=r_shoulder_pitch (NIB=0) (Errors:CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2158s 581m 262u : MC: generic motor error. (Joint=r_shoulder_roll (NIB=1) (Errors:CAN_RXWasPasv CAN_WasWarn ) 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2159s 359m 885u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2160s 361m 885u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2161s 363m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[DEBUG] yarprobotinterface running happily 
[WARNING] Provided non-increasing time vector 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2162s 365m 888u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2163s 367m 885u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ). 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2163s 582m 78u : MC: generic motor error. (Joint=r_shoulder_pitch (NIB=0) (Errors:CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2163s 582m 265u : MC: generic motor error. (Joint=r_shoulder_roll (NIB=1) (Errors:CAN_RXWasPasv CAN_WasWarn ) 
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2164s 39m 137u : MC: 2FOC quadrature encoder dirty. The number of thicks in a complete revolution of the motor was lower than expected, the optical disks need to be cleaned. (Joint=r_shoulder_pitch (NIB=0), Raw_quad_encoder_value=0) 
[... cut ...]
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=2184s 41m 142u : MC: 2FOC quadrature encoder dirty. The number of thicks in a complete revolution of the motor was lower than expected, the optical disks need to be cleaned. (Joint=r_shoulder_pitch (NIB=0), Raw_quad_encoder_value=0) 
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2184s 412m 885u : SYS: a service has detected that some CAN boards are not broacasting anymore. Type of service category is eomn_serv_category_inertials3. Lost can boards on (can1map, can2map) = ([ ], [ 13 ] ).
```

### Detailed context

Usually, it seems enough to idle and run again the motor from motorgui

AntonioConsilvio, 2024-12-17:
@S-Dafarra @illkid3 I have replaced the CAN termination on 2FOC 0B1 which could be one of the causes of this error, I ask for feedback when you can, Thank you!

AntonioConsilvio, 2025-01-24:
Hi! Do you know if this problem happened again? Thank you!

S-Dafarra, 2025-01-24:
Hi @AntonioConsilvio, after several days of usage, the error no longer appears. Thanks a lot! I am closing the issue

closed 2025-01-24 by S-Dafarra