robotology/mesh-support #1995: Left ankle roll temperature sensor broken

S-Dafarra, 2024-12-10:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

During Humanoids, the left ankle roll temperature sensor stopped working, preventing the robot to start successfully.

We tried replacing the i2c extender, but the 2foc went into short circuit and did not start at all.

### Detailed context

Unfortunately, we do not have any relevant log. In the end, we disabled the sensor.

### Additional context

cc @valegagge @MSECode @filippoborgogni

AntonioAzocar, 2024-12-16:
I replaced the TDB board of the ankle roll that was giving problems and after @AntonioConsilvio  tested it, it seems to be working again!

AntonioConsilvio, 2024-12-17:
Temperature sensor enabled:

- https://github.com/icub-tech-iit/robots-configuration/commit/7a0d5b61a882d782e8e167c9cbf56957f7fbac79

Good job @AntonioAzocar, Closing! ✅

closed 2024-12-17 by AntonioConsilvio