robotology/mesh-support #1996: Female pin of the fault button detached from the board

rob-mau, 2024-12-11:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The female pin to which the fault button cable connects became desoldered from the board after the cable was inadvertently pulled during the crane's movement. It was not possible to reconnect the cable.

### Detailed context

During the movement of the crane, I accidentally pulled out the male pin and The female pin was displaced within the torso cover. We then tried to reposition using a small screwdriver but it wasn't possible, and it was already detached. So we removed the back torso cover and realized the soldering was not intact. 
I left the cover disassembled and placed the female pin and all the screws in a small bag on the drawer unit at the entrance of the robot arena (picture attached).

### Additional context

[image]

[image]

SimoneMic, 2024-12-12:
I just want to mention that there are demos on Monday 16 Dec. and we need the robot to be operational, since also the other robots are in a bad shape.

Thank you!

cc. @AntonioConsilvio  @maggia80 @lornat75

AntonioConsilvio, 2024-12-13:
@edersonferrante replaced the broken board. The robot is working properly now!

closed 2024-12-13 by AntonioConsilvio