robotology/mesh-support #1997: Oscillations in left shoulder

rustlluk, 2024-12-12:
### Robot Name 🤖

iCubPrague01 S/N:047

### Request/Failure description

The left shoulder (mainly joints 1,2) of our robot starts to oscillate when the speed is about 10 deg/s and faster. 
@matejhof @SpisakJ 

### Detailed context

We initially thought it might be a problem only with the wrist, but we tried to idle the wrist, which helped slightly, but the shoulder still gets to oscillation. From the logs, we think it might be a problem with the PID for the shoulder. In the graphs and logs below, we sent the same commands to both arms, and only the left one started to oscillate (see also the video). When we set speed of 5deg/s for the shoulder joint, the problem does not occur.

We currently have the robot lying in a crib, hanged on steel ropes to support its weight. The problems with the arm were not present when the robot was standing in the usual pose.

### Additional context

Video:
https://github.com/user-attachments/assets/5006fe45-e0b5-4ea2-a65a-146d1e1601d6

Right Arm joint angles
[image]

Left Arm joint angle
[image]

Logs (joint angles + *Ext port)
[dataRightArmExt.log](https://github.com/user-attachments/files/18109076/dataRightArmExt.log)
[dataRightArm.log](https://github.com/user-attachments/files/18109077/dataRightArm.log)
[dataLeftArmExt.log](https://github.com/user-attachments/files/18109075/dataLeftArmExt.log)
[dataLeftArm.log](https://github.com/user-attachments/files/18109074/dataLeftArm.log)

pattacini, 2024-12-13:
Hi @rustlluk 

By inspecting the content of `dataRightArmExt.log`, I've checked that you're controlling the arms in standard position mode. By contrast, the corresponding log of the left arm indicates that the limb was already idle, if I'm not mistaken.

The PID gains are identical (barring the sign) for the left and the right arm, at least on our upstream repository:
- https://github.com/robotology/robots-configuration/blob/master/iCubPrague01/hardware/motorControl/left_arm-eb1-j0_3-mc.xml#L45-L58
- https://github.com/robotology/robots-configuration/blob/master/iCubPrague01/hardware/motorControl/left_arm-eb1-j0_3-mc.xml#L45-L58

Could you check on the robot system to see if this still holds true?

If this is the case, then it is likely that the left shoulder system is in a different state compared with the right one, meaning the mechanical response is different, probably due to some loose tendons, which may introduce elasticity hence a delay. In fact, by lowering the commanded average speed you obtain a stable behavior.

Possible actions:
1. Mechanical inspection to check the state of the tendons.
2. Detune the PID gains of the left shoulder.

For 2, you may be using [`robometry`][rob] to acquire the position setpoints as well as the feedback to attempt to run a system identification to help you tune the system.

[rob]: https://github.com/robotology/robometry

rustlluk, 2024-12-16:
Hello everybody,

Both arms should be controlled in position mode. Later, it actually happened that the left arm started to oscillate when going to the home position initiated from the MotorGUI, and I am sure all joints were in run mode (at least at the start of the movement).

I checked the config files for PID, and all values correspond to those in the repository. 

I have also tried to check the tendons, and the tension seems to be the same for both arms, but it is really hard to say. Is there any screw you would recommend to adjust?

Otherwise, I will try (hopefully this week) to tune the PID and will let you know.

Thanks!

pattacini, 2024-12-16:
>I have also tried to check the tendons, and the tension seems to be the same for both arms, but it is really hard to say. Is there any screw you would recommend to adjust?

Perhaps @AntonioConsilvio can provide a heads-up here.

---

Another possible source of problems may be the ETH connection with the board controlling the left shoulder. If it's not in a good state comm delays can arise, causing the oscillations.

You could ssh into icub-head and check the state with linux-based commands. The link must be up and running cleanly at 100 Mbps.

The correct IP can be retrieved from the XML files. For example:
- https://github.com/robotology/robots-configuration/blob/master/iCubPrague01/hardware/electronics/left_arm-eb1-j0_3-eln.xml#L11

rustlluk, 2024-12-18:
Just a heads up, we tried to put the robot back to the natural standing position. The left hand still shakes more than the right one, but there is no over-oscillation even with speeds of about 50deg/s. 

We will continue with the experiments next year and we will probably try to tune the PID a little bit.

AntonioConsilvio, 2025-01-28:
Hello @rustlluk, sorry for the late reply.

>I have also tried to check the tendons, and the tension seems to be the same for both arms, but it is really hard to say. Is there any screw you would recommend to adjust?

Judging by the behaviour there might be mechanical backlash on the joints of the left shoulder. From the video it is not easy to figure out where this backlash is coming from.

If the tendons are tense you can try starting the robot (`yarprobotinterface`) and manually moving the arm while the robot is controlled, trying to figure out which part of the shoulder is causing this backlash.

In the meantime, you can try tightening the following screws:

[image]

(under the upperarm cover)
[image]

As the problem occurs when the robot is lying down, the problem could be in the screws of the second photo.

Let us know how it goes when you are able to do these tests!

closed 2025-04-06 by github-actions[bot]