robotology/mesh-support #2001: Neck issue

ddetommaso, 2024-12-18:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

During the first run of the robot after returning from CRIS we found an issue on the neck. The neck does not move during calibration.

### Detailed context

From yarpmotorgui the motors are in hardware fault and there is no way to let them move. Moreover we found a cable disconnected (see attached pic). Manually moving the neck, the yaw seems loose.

### Additional context

[log_icub-head_yarprobotinterface_1886.txt](https://github.com/user-attachments/files/18184454/log_icub-head_yarprobotinterface_1886.txt)

### How does it affect you?

We need to start data collection for an experimental setting in January

filippoborgogni, 2025-01-14:
Hi @ddetommaso , tomorrow i will be in Erzelli to take a look to this issue.

@Fabrizio69 @AntonioConsilvio

filippoborgogni, 2025-01-16:
Hi @ddetommaso, yesterday I checked the robot and I solved the neck wiring problem. I tested the robot and everything works properly. Closing! 🚀

@Fabrizio69 @AntonioConsilvio

closed 2025-01-16 by filippoborgogni