robotology/mesh-support #2003: imu waist returning nan values for the orientation

GiulioRomualdi, 2024-12-20:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

⚠ I opened the issue here but I don't know if this is the right place. The problem might be related with the update done in the past few days (@traversaro @S-Dafarra)

I noticed that the IMU streams `nan` for the orientation.

### Detailed context

I tried to read the imu from the yarp port
```
yarp read ... /ergocub/waist/inertials/measures:o
```

This is the output
```
(((-7.18638353522891559777 4.72977836107371629737 1.20531494509568770646) 1734694961.05858850479)) (((-0.258610576391220148285 10.2859477996826171875 4.46157598495483487255) 1734694961.05858850479)) (((0.037499904632568359375 -0.106000185012817382812 -0.194000005722045898438) 1734694961.05445981026)) (((nan nan nan) 0.0)) () () () () () ()
```

### Additional context

I was not able to find error in the interface. Indeed the IMU seems to start as usual 
```
[INFO] Opening device waist-inertial with parameters [("robotName" = "ergoCubSN002"), ("serial" = "/dev/ttyS0"), ("xsensmt_period" = "0.01"), ("xsensmt_euler_period" = "0.005"), ("xsensmt_gyro_period" = "0.005"), ("xsensmt_acc_period" = "0.005"), ("xsensmt_mag_period" = "0.01"), ("sensor_name" = "waist_imu_0")]
[DEBUG] |yarp.dev.PolyDriver|waist-inertial| Parameters are (device xsensmt) (id waist-inertial) (robotName ergoCubSN002) (sensor_name waist_imu_0) (serial "/dev/ttyS0") (xsensmt_acc_period 0.00500000000000000010408) (xsensmt_euler_period 0.00500000000000000010408) (xsensmt_gyro_period 0.00500000000000000010408) (xsensmt_mag_period 0.0100000000000000002082) (xsensmt_period 0.0100000000000000002082)
[WARNING] xsensmt -  Parameter ' frame_name ' not set. Using default value ' waist_imu_0 ' for this parameter.
[INFO] xsensmt: Opening serial port /dev/ttyS0 with baud rate 115200. The following frequencies are considered. Accelerometer 200.0000 Hz, gyro 200.0000 Hz, magnetometer 100.0000 Hz, euler angle 200.0000 Hz, position 100.0000 Hz, linearVelocity 100.0000 Hz
[INFO] Device: MTi-630-8A1G6, with ID: 0080003FC8 opened.
[INFO] xsensmt: Putting device into configuration mode.
[INFO] xsensmt: Configuring the device MTi-630-8A1G6.
[INFO] xsensmt: Putting device into measurement mode.
[INFO] |yarp.dev.PolyDriver|waist-inertial| Created device <xsensmt>. See C++ class yarp::dev::XsensMT for documentation.
[INFO] Opening device waist-inertials_wrapper with parameters [("robotName" = "ergoCubSN002"), ("period" = "5"), ("name" = "/ergocub/waist/inertials")]
[DEBUG] |yarp.dev.PolyDriver|waist-inertials_wrapper| Parameters are (device multipleanalogsensorsserver) (id waist-inertials_wrapper) (name "/ergocub/waist/inertials") (period 5) (robotName ergoCubSN002)
[INFO] |yarp.device.MultipleAnalogSensorsServer| Parameter 'name' using value: /ergocub/waist/inertials
```

However here you can find the interface log
[imu_interface.txt](https://github.com/user-attachments/files/18210392/imu_interface.txt)

This is the status of the robotology
```
--------------------------------------------
/usr/local/src/robot/robotology-superbuild: master
2 weeks ago|Automatic update of latest releases versions (#1749)|4e9502df03f05fd32aeb69a0384a0b58e990e747|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/usr/local/src/robot/robotology-superbuild/src/BayesFilters/: master
1 year, 2 months ago|Merge pull request #103 from GiulioRomualdi/patch-1|b180d03df2614eeb7749e130c70391f59d91408c|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/usr/local/src/robot/robotology-superbuild/src/bipedal-locomotion-framework/: master
3 weeks ago|Estimate motor and joint velocities through KF (#909)|ba2a98aa93168374ece3f2913b15ee68bfaac6c4|HEAD -> master, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/usr/local/src/robot/robotology-superbuild/src/BlockFactory/: master
6 weeks ago|Merge pull request #88 from robotology/fixsimulinkbuild|825b6131a503254dd4c06abab04080de248368e1|HEAD -> master, tag: v1.0.1, origin/master, origin/HEAD
## master...origin/master
--------------------------------------------
/usr/local/src/robot/robotology-superbuild/src/casadi/: HEAD
3 months ago|Avoid dependency on distutils|a2d71bf95dbd3fa5f6725504bcba67dd52869d91|HEAD, tag: 3.6.7
## HEAD (no branch)
--------------------------------------------
/usr/local/src/robot/robotology-superbuild/src/CppAD/: HEAD
4 months ago|stable/20240000: check_version.sh: current master version set by new_release.sh.|67bdbf1bbf89cb0c490e7bdf6eac6fe92508f072|HEAD, tag: 20240000.7, origin/stable/20240000
## HEAD (no branch)
--------------------------------------------
[... 753 lines cut ...]
xorg-libxcomposite        0.4.6                h0b41bf4_1    conda-forge
xorg-libxdamage           1.1.5                h7f98852_1    conda-forge
xorg-libxdmcp             1.1.5                hb9d3cd8_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h4bc722e_1    conda-forge
xorg-libxinerama          1.1.5                h27087fc_0    conda-forge
xorg-libxrandr            1.5.2                h7f98852_1    conda-forge
xorg-libxrender           0.9.11               hd590300_0    conda-forge
xorg-libxt                1.3.0                hd590300_1    conda-forge
xorg-libxtst              1.2.5                h4bc722e_0    conda-forge
xorg-randrproto           1.5.0             hb9d3cd8_1002    conda-forge
xorg-recordproto          1.14.2            hb9d3cd8_1003    conda-forge
xorg-renderproto          0.11.1            hb9d3cd8_1003    conda-forge
xorg-util-macros          1.20.2               hb9d3cd8_0    conda-forge
xorg-xextproto            7.3.0             hb9d3cd8_1004    conda-forge
xorg-xf86vidmodeproto     2.3.1             hb9d3cd8_1005    conda-forge
xorg-xproto               7.0.31            hb9d3cd8_1008    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yarl                      1.18.0          py312h66e93f0_0    conda-forge
zeromq                    4.3.5                h3b0a872_7    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
```

S-Dafarra, 2024-12-20:
Just to mention a point of discussion with @GiulioRomualdi and @Giulero, maybe the fusion mode has not been activated on the IMU firmware

traversaro, 2024-12-20:
Thanks for the clear report. Can you check if containsEuler in https://github.com/robotology/yarp-device-xsensmt/blob/master/xsensmt/XsensMT.cpp#L123 is true or false?

GiulioRomualdi, 2025-01-08:
> Thanks for the clear report. Can you check if containsEuler in [robotology/yarp-device-xsensmt@`master`/xsensmt/XsensMT.cpp#L123](https://github.com/robotology/yarp-device-xsensmt/blob/master/xsensmt/XsensMT.cpp?rgh-link-date=2024-12-20T13%3A13%3A32.000Z#L123) is true or false?

Hi @traversaro sorry for the late answer. `packet->containsOrientation()` is false.

GiulioRomualdi, 2025-01-08:
Debugging on the robot I noticed that https://github.com/robotology/yarp-device-xsensmt/blob/bd3a04798977bda51c7a6e7ff55dcf3ab119bdfc/xsensmt/XsensMT.cpp#L311 is true so the euler should be set in the config

GiulioRomualdi, 2025-01-08:
If all agree, it is possible to unmount the IMU on SN002 and replace it with the one of the 001 that will be maintained and in the meantime, check the IMU's firmware to see if the fusion mode is disabled.

traversaro, 2025-01-08:
I double checked if for any reason the mt software vendored in the yarp-device-xsensmt was updated upstream, but I checked in https://www.movella.com/support/software-documentation and there the latest version is still 2022.2 .

GiulioRomualdi, 2025-01-08:
I was wondering if the fusion mode is disabled in the Movella GUI. I remember experimenting with one of those IMUs some time ago. It was possible to enable/disable the fusion mode and also change the sensor's location in terms of longitude and latitude to adjust the orientation of the magnetic field and the gravity vector. (Perhaps these parameters could also be added to the YARP device.)

maggia80, 2025-01-09:
> I was wondering if the fusion mode is disabled in the Movella GUI. I remember experimenting with one of those IMUs some time ago. It was possible to enable/disable the fusion mode and also change [...]

@AntonioConsilvio if you ask some of the raicam team could help check this, since they should have a cable to go directly to the sensor and have the Movella GUI SW installed on their computer.

AntonioConsilvio, 2025-01-13:
Hi @GiulioRomualdi! With the help of @kaidegast we were able to set up and test the sensor. 
It now seems to work correctly, but needs to be tested on the robot.

This is how we set up the sensor:

[image]

Please give me feedback when you are able to test the sensor!

GiulioRomualdi, 2025-01-13:
It works as expected! Thank you all

AntonioConsilvio, 2025-01-14:
Great! Closing ✅

closed 2025-01-14 by AntonioConsilvio