robotology/mesh-support #2004: Left shoulder yaw hard stop broken

GiulioRomualdi, 2024-12-20:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The left shoulder yaw hard stop is mechanically broken

### Detailed context

[image]

### How does it affect you?

As mentioned by @AntonioConsilvio, it is currently dangerous to start the robot.

Fabrizio69, 2024-12-20:
To better understand the problem, here is the broken mechanical part:

[image]

Cod.12279 - ICUB3_003_P_030

Fabrizio69, 2025-02-07:
The component cod.12279 - ICUB3_003_P_030 has been replaced