robotology/mesh-support #2006: `left_leg-eb8-j0_3` & `right_leg-eb6-j0_3` MC overheating and random `eo_transceiver_OccasionalROP_Load()` for `left_arm-eb2-j0_1`

SimoneMic, 2025-01-03:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

After a short time with the robot on, standing still on the ground, the `left_leg-eb8-j0_3` goes in overheating fault.
While keeping operating just the head I've noticed that also the right leg goes in overheating, although it's not shown on the motorGUI:

```
[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=2883s 847m 313u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_hip_pitch (NIB=0), Raw_temperature_value=-886)
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=2884s 273m 208u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_hip_pitch (NIB=0), Raw_temperature_value=363)
```

After some time with the robot standing still and operating only the head yaw joint, the motorGUI froze and some `eo_transceiver_OccasionalROP_Load()` errors started appearing . Crashing the walking-controller.

```
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=2884s 496m 595u : SYS: a service has detected that some CAN boards are still not transmitting. Type of service category is eomn_serv_category_battery. Lost CAN boards are on (can1map, can2map) = ([  ] , [ 2 ]). Total disappearance time: 2840240 [ms]
```

See detailed context for the full log

### Detailed context

```
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  20 12 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 8, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28159 32561 656
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  21119 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  10240 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 8, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  14336 32561 656
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  21504 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  10240 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  14336 32561 656
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 8, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  21504 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  10240 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  14336 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 8, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  21504 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  10240 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 6, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  14336 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 8, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  21504 32561 656
[... 89 lines cut ...]
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  0 32561 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 5, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  59392 32560 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 0, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  9216 32561 656
[ERROR] HostTransceiver::addSetROP__(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP 0.0.1.6 after all attempts with id:  [I, D, 3, 2,  , =,  , 0, x, 0, 1, 0, 0, 0, 1, 1, 4,  , -, >,  , I, N, D,  , =,  , 1, ,,  , T, A, G,  , =,  , e, o, p, r, o, t, _, t, a, g, _, m, c, _, j, o, i, n, t, _, c, m, m, n, d, s, _, s, e, t, p, o, i, n, t, 
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD right_leg-eb6-j0_3 IP 10.0.1.6 for nv ID32 = 0x01000114 -> IND = 1, TAG = eoprot_tag_mc_joint_cmmnds_setpoint
[ERROR] |yarp.devices.controlBoard_nws_yarp| Error while trying to command a streaming position direct message on joint group
[ERROR] HostTransceiver::addSetROP__(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP 0.0.1.5 after all attempts with id:  [I, D, 3, 2,  , =,  , 0, x, 0, 1, 0, 0, 0, 2, 1, 2,  , -, >,  , I, N, D,  , =,  , 2, ,,  , T, A, G,  , =,  , e, o, p, r, o, t, _, t, a, g, _, m, c, _, j, o, i, n, t, _, i, n, p, u, t, s, _, e, x, t, e, r, n, a, l, l, y, m, e, a, s, u, r, e, d, t, o, r, q, u, e, 
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD torso-eb5-j0_2 IP 10.0.1.5 for nv ID32 = 0x01000212 -> IND = 2, TAG = eoprot_tag_mc_joint_inputs_externallymeasuredtorque
[ERROR] HostTransceiver::addSetROP__(): ERROR in eo_transceiver_OccasionalROP_Load() for BOARD w/ IP 0.0.1.8 after all attempts with id:  [I, D, 3, 2,  , =,  , 0, x, 0, 1, 0, 0, 0, 2, 1, 4,  , -, >,  , I, N, D,  , =,  , 2, ,,  , T, A, G,  , =,  , e, o, p, r, o, t, _, t, a, g, _, m, c, _, j, o, i, n, t, _, c, m, m, n, d, s, _, s, e, t, p, o, i, n, t, 
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD left_leg-eb8-j0_3 IP 10.0.1.8 for nv ID32 = 0x01000214 -> IND = 2, TAG = eoprot_tag_mc_joint_cmmnds_setpoint
[ERROR] |yarp.devices.controlBoard_nws_yarp| Error while trying to command a streaming position direct message on joint group
[... cut ...]
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD torso-eb5-j0_2 IP 10.0.1.5 for nv ID32 = 0x01000114 -> IND = 1, TAG = eoprot_tag_mc_joint_cmmnds_setpoint
[ERROR] |yarp.devices.controlBoard_nws_yarp| Error while trying to command a streaming position direct message on joint group
[ERROR] theNVmanager::Impl::set() fails t->addSetROP() to BOARD right_arm-eb30-j4_6 IP 10.0.1.30 for nv ID32 = 0x01000114 -> IND = 1, TAG = eoprot_tag_mc_joint_cmmnds_setpoint
[ERROR] |yarp.devices.controlBoard_nws_yarp| Error while trying to command a streaming position direct message on joint group
```

mebbaid, 2025-01-07:
It happened again also in the right hip pitch (@S-Dafarra writing). By touching it did not seem very hot
```
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13365s 394m 626u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 305[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[INFO] from BOARD 10.0.1.9 (left_leg-eb9-j4_5) time=13366s 299m 51u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([  ], [ 1314 ])
[INFO] from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=13366s 382m 211u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([  ], [ 1314 ])
[INFO] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=13366s 284m 195u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([  ], [ 13 ])
[INFO] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13366s 426m 44u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([  ], [ 13 ])
[INFO] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=13366s 463m 501u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_ft. CAN boards are on (can1map, can2map) = ([  ], [ 13 ])
[WARNING] Provided non-increasing time vector
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=13367s 401m 107u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 341[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] Provided non-increasing time vector
[WARNING] Provided non-increasing time vector
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13368s 717m 621u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 305[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[INFO] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=13369s 11m 205u : SYS: a service has verified that the TX of its CAN boards is regular. Type of service category is eomn_serv_category_battery. CAN boards are on (can1map, can2map) = ([  ], [ 2 ])
[DEBUG] yarprobotinterface running happily
[WARNING] from BOARD 10.0.1.5 (torso-eb5-j0_2) time=13369s 546m 938u :  SYS: the DO phase of the control loop has last more than wanted. DO execution time 334[usec]. Latest previous execution times[usec] (..., Tx=N/A);(RX=123, DO=333, TX=135); (Rx=123, ...);
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=13372s 490m 386u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=r_hip_pitch (NIB=0), Raw_temperature_value=-886)
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=13372s 797m 429u : SYS: a service has detected that some CAN boards are still not transmitting. Type of service category is eomn_serv_category_battery. Lost CAN boards are on (can1map, can2map) = ([  ] , [ 2 ]). Total disappearance time: 13181082 [ms]
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13372s 892m 351u :  SYS: the TX phase of the control loop has last more than wanted. TX execution time 229[usec]. Latest previous execution times[usec] (RX=N/A, DO=N/A, TX=145); (Rx=80, DO=432, ...). Num of CAN frames[CAN1,CAN2]=[1 0]
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13372s 893m 769u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 453[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13372s 895m 724u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 413[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13372s 897m 636u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 321[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.5 (torso-eb5-j0_2) time=13374s 547m 938u :  SYS: the DO phase of the control loop has last more than wanted. DO execution time 332[usec]. Latest previous execution times[usec] (..., Tx=N/A);(RX=123, DO=332, TX=139); (Rx=127, ...);
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=13374s 669m 107u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 341[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13375s 195m 359u :  SYS: the TX phase of the control loop has last more than wanted. TX execution time 234[usec]. Latest previous execution times[usec] (RX=N/A, DO=N/A, TX=148); (Rx=80, DO=433, ...). Num of CAN frames[CAN1,CAN2]=[1 0]
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=13375s 197m 670u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 355[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[... cut ...] X=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=13543s 664m 133u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 369[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=13544s 150m 70u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 305[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
```

mebbaid, 2025-01-20:
This issue still persists on both left and right hip pitch with message

```bash
[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=3829s 278m 237u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_hip_pitch (NIB=0), Raw_temperature_value=358)
```

S-Dafarra, 2025-03-07:
Today the left hip pitch went into HF due to overheating and I was not able to recover it

S-Dafarra, 2025-03-07:
> Today the left hip pitch went into HF due to overheating and I was not able to recover it

In this case, the raw temperature value is -1141