robotology/mesh-support #2009: l_ring_pinky_oc does not move

S-Dafarra, 2025-01-08:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

After one or two teleoperation sessions, the left ring/pinky OC stopped working.

### Detailed context

The encoder seems to work, but the joint does not move, neither from motorgui

### Additional context

We did not do anything particularly stressful for the fingers, mostly movements without any load.

cc @fbiggi

mebbaid, 2025-01-14:
Sorry for the ping. Is there an update on this issue. This, and also #2010 are blocking for some demo scenarios required by a technology-transfer project.

edersonferrante, 2025-01-20:
Maintenance successfully completed
https://github.com/user-attachments/assets/de0ed61d-5bcb-42ea-9d26-499c501bdc50

Fabrizio69, 2025-01-22:
Hi @S-Dafarra, I think we can close this issue... 

cc: @edersonferrante @AntonioConsilvio

closed 2025-01-22 by Fabrizio69