robotology/mesh-support #2012: Torso pitch joint suffers frequent spikes

HosameldinMohamed, 2025-01-14:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

The torso pitch motor started to jump at random times, sometimes the jumps are large enough to stop the motor with a Hardware Fault.

Suddenly, it became so frequent.

### Detailed context

The Warning messages we found in yarp logger could be related to the issue

[image]

### Additional context

Looking at the robot we found the following wires cut from the same motor, not sure which sensor though:

[image]

CC @davidegorbani @pillai-s 

### How does it affect you?

This blocks experiments with iRonCub

HosameldinMohamed, 2025-01-16:
We unmounted the chest and back covers of the robot to look the connectors

[image]

From the motors' side the wires look fine

the other side of the cables is not accessible unless I unmount the 2FOC board then I can make a continuity test

HosameldinMohamed, 2025-01-16:
With @davidegorbani and @nicktrem we unmounted the 2FOC board `0B3` and were able to see the cable of the magnetic encoder WS1 going into the board, one or more of the wires are disconnected

[image]

filippoborgogni, 2025-01-20:
@AntonioConsilvio and I repaired the damaged cable. Please give us a feedback when you test it !

AntonioConsilvio, 2025-01-22:
Great! Closing 🚀

closed 2025-01-22 by AntonioConsilvio