robotology/mesh-support #2013: yarprobotinterface launched failed, possibly due to devices connection problem

jimingda, 2025-01-15:
### Robot Name 🤖

iCubShanghai01 S/N:049

### Request/Failure description

Hi,

I can launch the `yarprobotinterface` in the begining it turned green on `Yarpmanager`, but it failed after a while and errors as i showed in log below, then `yarprobotinterface` turned red on `Yarpmanager`'.

After icub@icub-head:~$ `FirmwareUpdater -a`, then `Discover` the ETH, i got the following result.

[image]

[log_icub-head_yarprobotinterface_1698.txt](https://github.com/user-attachments/files/18419164/log_icub-head_yarprobotinterface_1698.txt)

### Detailed context

None

### Additional context

None

jimingda, 2025-02-28:
Hi,

Hope this message finds you well. Could someone please take a moment to review this issue?

I've been struggling with this problem for quite some time now, and any assistance would be greatly appreciated.

Thank you very much in advance!

jimingda, 2025-03-10:
Hi, @AntonioConsilvio 

Sorry to bother you, can you help me to have a look at this problem, thank you.

AntonioConsilvio, 2025-03-10:
Hi @jimingda, sorry for the long wait.

You were right to bring this to our attention. We will make your robot issues a priority and get you support as soon as possible. 
You will be updated shortly.

filippoborgogni, 2025-03-13:
Hi @jimingda, 
looking at the problems described in the issue and as can be seen from `FirmwareUpdater` there is an ethernet connection problem between the EMS4 board (EB1) and the MC4PLUS (EB24). 

[image]

You should check the status of the cable in question that is called E1 and runs from P3 on the EB1 (`10.0.1.1`) to P13 (circled in red🔴 in the photo below) on the EB24  (`10.0.1.24`). 

[image]

In the photo below you can see the path of the cable (in blue 🔵) which starts at the EMS4 (in red 🔴), passes through the sheath of the left shoulder and arrives at the MC4PLUS (in yellow 🟡)

[image]

For better clarity I also enclose a photo, taken from the M&B placement, where you can see that the EB1 is below the 2FOC and that the EB24 of the 3 MC4PLUS is the one most attached to the arm mechanics.

[image]

[image]

Let us know if this information was useful and if you can solve the problem!

cc @AntonioConsilvio