robotology/mesh-support #2015: Microphone and Speaker not working

giotherobot, 2025-01-15:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

Both the microphone and speaker of the robot are not working.

### Detailed context

Both the microphone and speaker of the robot are not working.

filippoborgogni, 2025-01-20:
Hi @giotherobot, @AntonioConsilvio and I found the cable disconnected and we repaired it. Let us know if it works properly !

giotherobot, 2025-01-21:
I am testing the audio in/out and it does not seem to work yet.

For the speaker, in the `robot-head` I am testing with: `test-speaker`
I do not hear anything.
Opening alsamixer shows 3 sound cards:

```
│ 0 [A              ]: USB-Audio - USB-C to 3.5mm Headphone Jack A                                 │
│                      Apple, Inc. USB-C to 3.5mm Headphone Jack A at usb-3610000.usb-3.3, full spe│
│ 1 [HDA            ]: tegra-hda - NVIDIA Jetson Orin NX HDA                                       │
│                      NVIDIA Jetson Orin NX HDA at 0x3518000 irq 120                              │
│ 2 [APE            ]: tegra-ape - NVIDIA Jetson Orin NX APE                                       │
│                      NVIDIA-CTIBosonOrinNX-NotSpecified-Jetson 
```

The looking in /etc/asound.conf:

```
cat /etc/asound.conf
#
#  ALSA library configuration file
#
#  SPDX-FileCopyrightText: Copyright (c) 2018-2021 NVIDIA CORPORATION & AFFILIATES. All rights reserved.
#  SPDX-License-Identifier: LicenseRef-NvidiaProprietary
#
# NVIDIA CORPORATION, its affiliates and licensors retain all intellectual
# property and proprietary rights in and to this material, related
# documentation and any modifications thereto. Any use, reproduction,
# disclosure or distribution of this material and related documentation
# without an express license agreement from NVIDIA CORPORATION or
# its affiliates is strictly prohibited.

pcm.!default {
	type plug
	slave {
		pcm "hw:APE,0"
		channels 2
		rate 48000
	}
	hint.description "Tegra APE Soundcard"
}

ctl.!default {
	type hw
	card APE
}

pcm.demixer {
	type plug
	slave {
		pcm "dmix:APE,0"
		channels 2
		rate 48000
	}
}

```
It seems that the default sound card is not the USB adapter. Looking into how to change the default.

Instead after increasing the `Mic Boost` slider in `alsamixer` on the `ergocub-torso` the microphone seems to be working (`test-microphone`).

giotherobot, 2025-01-21:
Upon reboot the sound card default was back to the tegra one. Investigating.

giotherobot, 2025-01-21:
Using the info in https://askubuntu.com/questions/980991/pulseaudio-does-not-remember-default-sink I edit the `/etc/pulse/default.pa` file and added this line:
```
set-default-sink alsa_output.usb-Apple__Inc._USB-C_to_3.5mm_Headphone_Jack_Adapter_DWH41830A9GJKLTA5-00.analog-stereo
```
This works after running `test-speaker`, but before running it there is no pulseaudio loaded so running `pacmd list-sinks | grep "index:\|name:"` was not enough to check if the default was changed ( it resulted in `No PulseAudio daemon running, or not running as session daemon.`). 

Anyway now it works after reboot.

closed 2025-01-21 by giotherobot