robotology/mesh-support #2017: Right hand boards issue

ddetommaso, 2025-01-17:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

Even if the calibration seems working we have an issue with some boards of the right hand.

### Detailed context

There is probably an issue with the thumb oppose

[image]

### Additional context

[log_icub-head_yarprobotinterface_2128.txt](https://github.com/user-attachments/files/18457159/log_icub-head_yarprobotinterface_2128.txt)

### How does it affect you?

CRITICAL! We need to start a data collection for an experimental study in a few days.

AntonioConsilvio, 2025-01-21:
Hi @ddetommaso, @filippoborgogni will check this issue tomorrow!

filippoborgogni, 2025-01-23:
Hi @ddetommaso , yesterday I found a short circuit on the MAIS board connected to the hall sensor of the right thumb. Unfortunately I was not being able to repair it, so I detached the sensor of the right thumb and disabled the finger. In this moment all fingers are working expect for the thumb. During the test the proximal tendon of the thumb breaks and I fixed it. We will come back in the next period, but, talking with @mmwozniak, you should be able to do your experiments.

ddetommaso, 2025-01-23:
Hi @filippoborgogni,

thank you for taking care of this. At the moment we are not sure we will be able to start with the experiments because we are not able to establish whether the subsequent repair of the robot will affect the execution times of the movements involving the arms. It is a risk that we will evaluate in the coming days. In any case, my request is to fix the problems on this robot as soon as possible.

@mmwozniak @AntonioConsilvio

AntonioConsilvio, 2025-01-23:
OK @ddetommaso, then please let us know how the robot's status will affect the next experiments!

ddetommaso, 2025-01-28:
@AntonioConsilvio the issue seems not blocking for us for running experiments. Thus, the robot will not be available for fixing the next following weeks.

ddetommaso, 2025-02-26:
@AntonioConsilvio could you please schedule an intervention on this robot around the 19th of March? We would need the robot fully working by the 21st of March.

Thank you

AntonioConsilvio, 2025-03-04:
We have scheduled the intervention for Wednesday 12 March!

filippoborgogni, 2025-03-13:
Hi @ddetommaso , yesterday @AntonioConsilvio and I worked to resolve this issue. We re-soldered several cables of the right thumb hall sensors and the short circuit previously found on the MAIS board seems solved.
During the intervention, the proximal tendon of the right thumb also broke, which we replaced. 
Unfortunately, one of the two screws that block the tendon on the side of the finger was stripped and, after trying to extract it without success, we blocked the tendon with a double knot. 

[image]

In the next period (and when we have the mechanical part available) we will come back to replace the piece, but in any case the robot is currently fully functional (as you can see in the videos below). 

https://github.com/user-attachments/assets/2fdd813f-7331-4bfc-af3e-4be7bd053f30

https://github.com/user-attachments/assets/51ab20ca-3dc0-4c7f-b9e3-e78411e9903c

We were also able to observe a mechanical backlash in the neck which is much more evident when the motors are not controlled. 
A new [issue ](https://github.com/robotology/icub-tech-support/issues/2059)has been opened for this problem, so I'd say this one can be closed! 🚀

closed 2025-03-13 by filippoborgogni