robotology/mesh-support #2026: Right thumb adduction encoder stuck

S-Dafarra, 2025-02-04:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

While using the robot, we noticed that the right thumb adduction joint was going to the limit. Looking at the motorgui, we noticed that the encoder value was stuck 

### Detailed context

By moving the joint manually when in idle, the encoder value does not seem to change

Fabrizio69, 2025-02-10:
Hi @S-Dafarra , can we consider this problem solved? It was the FAP connected to the wrong connector of the right hand MMA board. During the ergoCub hand test meeting we fixed it. Do you confirm?

cc: @AntonioConsilvio

S-Dafarra, 2025-02-10:
Yes it is fixed, thanks!

closed 2025-02-10 by AntonioConsilvio