robotology/mesh-support #2029: Robot does not start for left ankle roll  overheating

Giulero, 2025-02-05:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

Similar to https://github.com/robotology/icub-tech-support/issues/2016 When starting the robotinterface we got several errors due to overheating. 

### Detailed context

This is the error
```
[ERROR] from BOARD 10.0.1.9 (left_leg-eb9-j4_5) time=104s 879m 318u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_ankle_roll (NIB=1), Raw_temperature_value=-886)
```

### Additional context

Last time it was a damaged cable

valegagge, 2025-02-06:
Hi @AntonioConsilvio ,
could you tell me if there is the i2c_ext board on the affected joint, please?
thanks

S-Dafarra, 2025-02-07:
It is happening quite often:
- https://github.com/robotology/icub-tech-support/issues/2016
- https://github.com/robotology/icub-tech-support/issues/1995

maggia80, 2025-02-07:
@S-Dafarra do you have a log of the temperature? Because the problem could be a real overheating.

@MSECode could you help with the log in case we don't have any?

MSECode, 2025-02-07:
Actually, looking at the error message on the log the problem was due to the fact that the TDB stopped receiving data on the i2C bus (raw value -886). Now, this might be due to the following reasons:
- cable or connector disconnected or partially out from the crimp, thus in some position it does not make good contact
- sensor that froze up and was not able to receive the restart request --> this problem is not totally clear and might be investigated more. As a matter of fact in other similar issues happened that when restarted the sensor was able to read again. Thus, we are not sure if this could be due to a sensor "freezing" or to the cable. 

See the issues linked by stefano at the comment: https://github.com/robotology/icub-tech-support/issues/2029#issuecomment-2642362666 where we basically had the same exact problem. In that case was due to the cable outside of the crimp.

valegagge, 2025-02-10:
I guys, 

in order to keep tack of the issues related to the motor temperature I started to collect all the last issues (after the use of the I2c-EXT boards) in a table reporting a;also the cause of the raised problem.

Here it is: https://github.com/icub-tech-iit/five/issues/291

For whom hasn't the access, here I report the table I created. 

| Issue                                                                                                                      | Link                                                        | Cause                                                                                                                                                                                                                                                                                   | Solution                                        |
|----------------------------------------------------------------------------------------------------------------------------|-------------------------------------------------------------|-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|-------------------------------------------------|
| ergoCub 1.3 S/N:002 – Robot does not start for left ankle roll overheating                                                 | https://github.com/robotology/icub-tech-support/issues/2016 | `[ERROR] from BOARD 10.0.1.9 (left_leg-eb9-j4_5) time=42s 274m 261u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_ankle_roll (NIB=1), Raw_temperature_value=-886)`                   | Cable replaced                                  |
| ergoCub 1.3 S/N:002 – l_shoulder_pitch goes into overheating fault                                                         | https://github.com/robotology/icub-tech-support/issues/2007 | ` 	195.915567	ERROR			from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=550s 816m 404u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_shoulder_pitch (NIB=0), Raw_temperature_value=-886)` | Cable repaired                                  |
| ergoCub 1.3 S/N:002 – left_leg-eb8-j0_3 & right_leg-eb6-j0_3 MC overheating and random eo_transceiver_OccasionalROP_Load() | https://github.com/robotology/icub-tech-support/issues/2006 | `[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=2883s 847m 313u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_hip_pitch (NIB=0), Raw_temperature_value=-886)`                  |  Still open                                     |
| ergoCub 1.1 S/N:001 – Right knee vibrates while calibration                                                                | https://github.com/robotology/icub-tech-support/issues/2000 |                                                                                                                                                                                                                                                                                         | fix tendeon and temperature calble of hip pitch |
| ergoCub 1.3 S/N:002 – Left ankle roll temperature sensor brokensupport/issues/1995                                         | https://github.com/robotology/icub-tech-support/issues/1995 | left ankle roll temperature sensor stopp. We tried replacing the i2c extender, but the 2foc went into short circuit and did not start at all.ed working                                                                                                                                 | TDB board replaced                              |
| ergoCub 1.3 S/N:002 – Weird smell coming from the robot and errors on the right leg                                        | https://github.com/robotology/icub-tech-support/issues/1987 | A few minutes after startup we started smelling something weird coming from the robot. From the motorgui I noticed that both right ankle roll and pitch were in hardware fault.                                                                                                         | motor replaced. Cuyrrent fw issue               |
| ergoCub 1.3 S/N:002 – Issue on right ankle pitch temperature sensor                                                        | https://github.com/robotology/icub-tech-support/issues/1986 | high temperature value                                                                                                                                                                                                                                                                  |  motor was burnt out                            |
|  ergoCub 1.1 S/N:001 – [Fake overheating motor l_knee]                                                                     | https://github.com/robotology/icub-tech-support/issues/1941 | `[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=154s 378m 40u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_knee (NIB=3), Raw_temperature_value=385)`                          | cable repaired                                  |

mebbaid, 2025-03-05:
perhaps we can add this one as well https://github.com/robotology/icub-tech-support/issues/2051