robotology/mesh-support #2032: Left thumb OC does not close fully

S-Dafarra, 2025-02-13:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

The left thumb OC seems to close only up to 55 deg

### Detailed context

We were doing some teleoperation experiments, and then it stopped moving. After rebooting, it was reaching only 55 deg

S-Dafarra, 2025-02-17:
I have updated the title and the description of the issue with the list of fingers that have limited range of motion.

cc @Fabrizio69 @maggia80

S-Dafarra, 2025-02-18:
After using the robot a bit, now the left ring finger closes only up to 60deg, and very slowly

S-Dafarra, 2025-02-24:
Now the left thumb range is basically none: #2039

maggia80, 2025-02-25:
@AntonioConsilvio @filippoborgogni could you check the motor position in the motorgui? I want to be sure the motors reach the max position. 
 `<param name="rotorPosMin">          -26500         -26500          -26500          -26500      </param> `

S-Dafarra, 2025-02-28:
The left pinky finally gave up: https://github.com/robotology/icub-tech-support/issues/2048

S-Dafarra, 2025-02-28:
An update on the other fingers.
- Left thumb OC closes only of **20deg** and very slowly
- Left middle OC closes up to 60 deg
- Right middle OC reaches 60deg
- Right pinky OC reaches 70deg

It seems that we loose more or less 15 deg per week in each finger (except the thumb) and after three weeks the tendon snaps

edersonferrante, 2025-03-06:
Hi @S-Dafarra, probably the problem of decreasing grades is due to the tendon pulling out that the more it pulls out the more grades it loses and the slower is the closing of the fingers, also because as soon as you change tendon with a new one this problem is solved automatically.
   
 [image]
@Fabrizio69 @maggia80

edersonferrante, 2025-03-12:
Hi @S-Dafarra, before removing the tendon of the little finger and ring finger and redoing it I took some photos to see if there was anything strange, but I would say that I found nothing.

[image]

cc @maggia80

AntonioConsilvio, 2025-03-18:
Hi! We performed a mechanical upgrade for both lower arms. Now all fingers are working properly.

I will close all issues related to the fingers to make sure that the next problems happen after the upgrade.

closed 2025-03-18 by AntonioConsilvio