robotology/mesh-support #2033: Sometimes the fingers stop responding until a full reboot

S-Dafarra, 2025-02-13:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

It happens often that while performing some motions with the fingers, some of them stop moving.

Even by changing the reference from motorgui the finger stays stuck. Apart from the tendon issues, sometimes it seems enough to simply reboot the robot.

### Detailed context

This behavior was noticed also by @AntonioConsilvio, with some of the fingers' boards shutting down during usage, but I wanted to open an issue to keep track of this. 

This is happening on most of the fingers. The only joint I am not certain of is the thumb adduction.

cc @maggia80

maggia80, 2025-02-13:
Cc @marcoaccame @fiorisi

AntonioConsilvio, 2025-02-14:
Hi @S-Dafarra, is it possible to get a picture of the `yarpmotorgui` at the time when one or more fingers are closed, showing this problem?

AntonioConsilvio, 2025-03-19:
Closing due this comment:

- https://github.com/robotology/icub-tech-support/issues/2032#issuecomment-2732197947

closed 2025-03-19 by AntonioConsilvio