robotology/mesh-support #2034: Left foot FT mostly saturated on one channel

S-Dafarra, 2025-02-17:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

By checking the FirmwareUpdater, we noticed that one channel outputs 24000 (having a 32000 range) with the robot hanging on the crane, hence with a very small weight attached to it.

[image]

### Detailed context

We noticed that the ``yarprobotinterface`` was throwing errors about a saturated FT while walking. The errors are not continuous, nonetheless, the performance of the robot while walking are not very good. It shakes more.

cc @DanielePucci @Giulero @GiulioRomualdi @maggia80

S-Dafarra, 2025-02-20:
When the robot is on the ground it is enough to shake it a little to see saturations coming up on the interface.

AntonioConsilvio, 2025-02-21:
Hi @S-Dafarra, we will plan the replacement of the FT sensors soon

edersonferrante, 2025-02-24:
Hi @S-Dafarra , we have replaced FT S/N 133 with FT S/N 79, we are waiting for feedback.

S-Dafarra, 2025-02-24:
Hi @edersonferrante, now when the robot is on the ground, I don't see saturation errors anymore!

AntonioConsilvio, 2025-02-25:
Good job! Closing! 🚀