robotology/mesh-support #2035: Right index OC broken

S-Dafarra, 2025-02-18:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

While trying to handle some boxes during teleoperation, we heard a crack and after that the right index was not closing anymore

### Detailed context

The motorgui shows a high number for the encoder, even if the finger is straight

[image]

fbiggi, 2025-02-20:
[image]

[image]

[image]

[image]

[image]

S-Dafarra, 2025-02-20:
For reference, this is how it got broken

https://github.com/user-attachments/assets/8ae0526d-4ec0-43f4-8b0c-bae18b1b5cc3

AntonioConsilvio, 2025-02-27:
Finger calibrated and tested. Closing! 🚀