robotology/mesh-support #2036: Left thumb adduction mechanically broken

S-Dafarra, 2025-02-18:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

While manipulating a box, the robot fingers got a little bit stuck in the handles. After that we heard a crack and then we heard the finger motor spinning. The finger itself instead is very loose.

### Detailed context

I believe that the motor axis broke.

cc @Fabrizio69 @maggia80 @fiorisi

fbiggi, 2025-02-19:
@AntonioConsilvio 
The thumb adduction motor has been replaced
I attach the [Proto issue](https://github.com/icub-tech-iit/proto/issues/3165) of the work performed

@S-Dafarra @DanielePucci @maggia80

S-Dafarra, 2025-02-19:
For reference, this is how it got broken

https://github.com/user-attachments/assets/67553c76-40bb-4968-a082-b8b748df7aa8

I wonder why it got broken instead of going into hardware fault

edersonferrante, 2025-02-27:
Hi @S-Dafarra, we have tried to eliminate as much play as possible, but we believe the issue is due to the deformation of the mechanical part (Code 16674, Rev. A1, and Alias IC_028_P_005) shown in the figure. We will need to disassemble the mechanical part and conduct a test to verify if it has indeed deformed, If that is the case we will replace it with a new one.
As soon as possible we will proceed with the test. 

[image]

@Fabrizio69

AntonioConsilvio, 2025-03-18:
Hi @S-Dafarra, We have temporarily solved this problem by using a little drop of glue to hold the mechanism in place, and now the adduction is working properly. 

We are keeping in mind that should any problems arise on that part of the mechanism, it will most likely be necessary to replace the entire motor assembly.

cc @fbiggi @edersonferrante