robotology/mesh-support #2038: Left wrist M1 belt completely worn down

S-Dafarra, 2025-02-24:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The belt running on the M1 motor is completely worn down, and about to break, while the belt tensioner has a sign on it.

### Detailed context

After about one hour of teleoperation, we noticed that the left wrist was in a completely different position than the right wrist.

@AntonioConsilvio has a picture of it

### Additional context

cc @maggia80 @fiorisi @Fabrizio69

AntonioConsilvio, 2025-02-25:
As shown in this picture, the belt is completely damaged (red circle), and the belt tensioner is scored (purple circle).

[image]

The possible reason is the belt tensioner screw, in fact from this picture you can see that it is very close to the brass belt tensioner, instead of touching the bearing that holds the belt tensioner (see difference with CAD picture).

[image]

[image]

This damages the belt tensioner and affects the belt. We will plan the replacement soon.

cc @maggia80 @Fabrizio69 @fiorisi @fbiggi

Lawproto, 2025-02-25:
thanks @AntonioConsilvio, by design that should not occur. Can you tell me if the spacer ring was implemented in the physical robot before the damage?

[image]

Fabrizio69, 2025-02-28:
The spacer ring (Cod.4591-HARWIN_R30-6700394) is present in the ergocub_sn_001 snapshot, so it should be present on the robot:

[image]

@edersonferrante, can you check, please?

Gandoo, 2025-03-06:
ciao, the forearm sx was reassembled on the robot and turned on by @isorrentino to check the connections. 
 **I remind everyone that @AntonioConsilvio has not yet done the joint calibration.**

[image]

[image]

cc @S-Dafarra @fbiggi @DanielePucci @maggia80  @GiulioRomualdi @Fabrizio69