robotology/mesh-support #2040: Right thumb fingertip detached

S-Dafarra, 2025-02-25:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

The right thumb fingertip is falling apart.

For the time being, we locked it in place with some tape

### Detailed context

[image]

Fabrizio69, 2025-02-28:
Hi @edersonferrante, can you check this issue?

edersonferrante, 2025-02-28:
Hi @Fabrizio69 , of course, I’ll see if the problem can be solved.

AntonioConsilvio, 2025-03-03:
cc @simeonedussoni @fiorisi

AntonioConsilvio, 2025-03-18:
Hi! We replaced all the phalanxes of the hands for an upgrade, and now the fingertips are all firmly in place. Let's see if the problem recurs in the future.

Closing!

closed 2025-03-18 by AntonioConsilvio