robotology/mesh-support #2041: Does not start after firmware and  robotology update

isorrentino, 2025-02-25:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

We tried to update the firmware to the latest devel using the script. The robot doesn't start anymore and we cannot see the amc bord from the firmware updater 

### Detailed context

[image]

### Additional context

The robot doesn't start with this error 

[error_interface.txt](https://github.com/user-attachments/files/18969882/error_interface.txt)

marcoaccame, 2025-02-25:
from reading the log:

boards ok: .22, .23, .24, .25, 

at a certain point the receiver in YRI tells that it has lost about 2K packets from boards .22. and .24

[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 2, .,  , E, x, p, e, c, t, e, d, :,  , 3, 1, 0, 7, ,,  , R, e, c, e, i, v, e, d, :,  , 5, 1, 4, 9, ,,  , M, i, s, s, i, n, g, :,  , 2, 0, 4, 2, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 4, 7, 5, 2, 4, 0, 4, 0,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 4, 9, 5, 6, 7, 0, 4, 0,  , u, s, 
[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 4, .,  , E, x, p, e, c, t, e, d, :,  , 2, 5, 9, 8, ,,  , R, e, c, e, i, v, e, d, :,  , 4, 6, 4, 1, ,,  , M, i, s, s, i, n, g, :,  , 2, 0, 4, 3, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 4, 7, 5, 1, 9, 4, 3, 9,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 4, 9, 5, 6, 3, 4, 3, 9,  , u, s, 

then it tells that these two boards have shown up after about 2 seconds (that with 1 frame every 1 ms is about 2000 frames)
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.22 (right_arm-eb22-j7_10) has shown after being lost for 2.04301 sec
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.24 (right_arm-eb24-j11_12) has shown after being lost for 2.04402 sec

**that is a bad sign**. maybe crc errors or link down. later on i will look for it

**BINGO**: we have a link that has been down in input to board .22:
[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=49s 635m 159u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.

and we have very many similar errors further on for the same board. there must be a loose cable:

[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=54s 935m 159u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
...
[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=60s 135m 160u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
...
[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=65s 435m 160u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
...
[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=70s 735m 160u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
...
[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=75s 935m 160u :  ETH monitor: link goes up. in port ETH input (P2/P13/J4). Application state is unknown.
...

the above errors of link down happen when YRI attempts to establish conenction w/ board 10.0.1.2 but its attempts do no reach the board because of the link down
between boards .22 and its previous. 

aventually YRI gives up, produces error:
[ERROR] EthResource::verifyBoardPresence() DID NOT have replies from BOARD left_arm-eb2-j0_1 with IP 10.0.1.2 after 21.003 seconds: CANNOT PROCEED ANY FURTHER

and forces a shutdown

**REMEDY**:
at first verify ETH cable in input to board 10.0.1.22 and reattempt the launch of YRI

marcoaccame, 2025-02-25:
There are other problems:

1. the mc4plus boards seems to have a FW version of July 2024 ver 3.94 and now devel has v 3.101
 ```
[INFO] EthResource::askBoardVersion() found BOARD head-eb20-j0_1 @ IP 10.0.1.20 of type mc4plus with FW = ver 3.94 built on 2024 Jul 04 13:42
```
the old v 3.94 has an incompatible protocol w/ latest icun-firmware-build that cause a failure in setting the  configuration of the motor:
```
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 1
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 2
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 3
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 4
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 5
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 6
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 7
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 8
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 9
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 10
[WARNING] theNVmanager::Impl::setcheck() had an error while calling check() in BOARD head-eb20-j0_1 with IP 10.0.1.20 at attempt # 11
[ERROR] FATAL: theNVmanager::Impl::setcheck() could not set and verify ID ID32 = 0x01010001 -> IND = 0, TAG = eoprot_tag_mc_motor_config in BOARD head-eb20-j0_1 with IP 10.0.1.20  even after  11 attempts
[ERROR] FATAL: embObjMotionControl::init() had an error while calling setcheckRemoteValue() for motor config fisico # 0 in  BOARD head-eb20-j0_1 (IP 10.0.1.20) 
```

**REMEDY**: update the mc4plus boards

2. there is a wrong application protocol in the xml file fro the amcbldc / amc2c below board 10.0.1.31. the devel version is 2.1.0 / 3.4.0 and teh detected is ver 2.1.0 / 3.4.0 (SO OK). but the xml file wanted the 2.0.13 / 3.0.3
```
[INFO] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=75s 879m 774u :  CFG: CANdiscovery started the search for 2 eobrd_amcbldc boards on (can1map, can2map) = (0x0006, 0x0000) with target can protocol ver 2.0 and application ver 2.0.13.
[INFO] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=75s 879m 848u :  CFG: CANdiscovery started the search for 1 eobrd_amc2c boards on (can1map, can2map) = (0x0008, 0x0000) with target can protocol ver 2.0 and application ver 3.0.3.
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=75s 880m 550u : CFG: CANdiscovery has detected  eobrd_amcbldc board in CAN1 addr 1 with can protocol ver 2.0 and application ver 2.1.0 Search time was 0 ms
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=75s 880m 623u : CFG: CANdiscovery has detected  eobrd_amcbldc board in CAN1 addr 2 with can protocol ver 2.0 and application ver 2.1.0 Search time was 0 ms
[WARNING] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=75s 880m 695u : CFG: CANdiscovery has detected  eobrd_amc2c board in CAN1 addr 3 with can protocol ver 2.0 and application ver 3.4.0 Search time was 0 ms
[ERROR] from BOARD 10.0.1.31 (left_arm-eb31-j4_6) time=75s 880m 769u : CFG: CANdiscovery detected 2 invalid eobrd_amcbldc boards in CAN1:
	 1 of 2: wrong eobrd_amcbldc because it has: WRONG APPLICATION VERSION 
	 2 of 2: wrong eobrd_amcbldc because it has: WRONG APPLICATION VERSION 
```

**REMEDY**: review the xml file

traversaro, 2025-02-25:
> **REMEDY**: review the xml file

Probably we need to propagate the ergoCubSN001 the changes done in ergoCubSN002 in https://github.com/robotology/robots-configuration/commit/84d22ac76988dbcfc08dea30edad6b48df31eeca .

traversaro, 2025-02-25:
> the mc4plus boards seems to have a FW version of July 2024 ver 3.94 and now devel has v 3.101

This is strange, probably the automated firmware update failed? @isorrentino @GiulioRomualdi did you saved the log of the update script (something like https://github.com/robotology/icub-firmware-build/issues/176#issue-2714724526)? Did the firmware update ended with success or failure?

traversaro, 2025-02-25:
> > **REMEDY**: review the xml file
> 
> Probably we need to propagate the ergoCubSN001 the changes done in ergoCubSN002 in [robotology/robots-configuration@84d22ac](https://github.com/robotology/robots-configuration/commit/84d22ac76988dbcfc08dea30edad6b48df31eeca) .

PR proposed in https://github.com/icub-tech-iit/robots-configuration/pull/4 .

isorrentino, 2025-02-25:
> This is strange, probably the automated firmware update failed? @isorrentino

We did not save the log but I'm quite sure the script updated the ems boards correctly as I checked the version and it was 3.101 for all the boards.
The problem is related to the configuration files which still have the old version. But if I remember correctly, at a certain point, there has been a shared decision for no longer checking the version. Am I wrong? @AntonioConsilvio.
Anyway this error related the fw versions should be easy to fix.

isorrentino, 2025-02-25:
> Interestingly, the 10.0.1.22 board seems to be visibile in the screenshot of the original post, perhaps it could make sense to manually update the latest firmware?

Is it possible that the problem is  not hardware? After the demo of this morning no one turned the robot on, and we started the update right after. I don't know if this error can be related to a connection issue.

traversaro, 2025-02-25:
> > This is strange, probably the automated firmware update failed? [@isorrentino](https://github.com/isorrentino)
> 
> We did not save the log but I'm quite sure the script updated the ems boards correctly as I checked the version and it was 3.101 for all the boards.

Just to clarify, the board that seems to have a wrong firmware version is a `mc4plus`, so it is possible that the script updated all the `ems` correctly, but not all the `mc4plus`.

isorrentino, 2025-02-26:
Got it!

Fabrizio69, 2025-02-26:
@AntonioConsilvio is checking the electrical connections...

AntonioConsilvio, 2025-02-26:
I try to shed some light on how we fixed the problems.

The application update of several boards at the same time can sometimes result in errors that can lead to the loss of the application firmware (not only on this robot and not only on ergoCub).

This happens not only using the script, but can also happen with manual updating of boards using the `FirmwareUpdater` (rarer, but it happens).

Two possibilities arise at this point:

1. The board stays in `eUpdater` (see `10.0.1.3` board in the main comment picture).

2. The board "disappear", i.e. it is no longer displayed by the `FirmwareUpdater` (see `10.0.1.2` and `10.0.1.31` boards in the main comment picture) because it doesn't pass the bootloader part (which takes about 5/10sec) and so is not in either `eApplication` or `eUpdater`.

Both problems are solved:

1. Just start `FirmwareUpdater -a`, reload the application firmware and select the board and click on `Set Def Boot eApplication` (1) and `Force ETH Application` (2).

[image]

2. This is solved as in problem 1, but with one more step, which is to restart the motors (power-cycle) do the `discover` of the ETH boards within 5/10 secs of boot. At this point you will see all the ETH cards in `eUpdater` (including the ones that were `disappeared`). Then reload the firmware as in problem 1.

So in conclusion there were no hardware problems and now the robot starts properly.

The problem could also happen with `CAN` boards, the resolution method is almost the same.

ps. The `Set Def Boot eApplication` part may not be strictly necessary, but I do it just in case.

cc @isorrentino @GiulioRomualdi @traversaro @marcoaccame @Fabrizio69

pattacini, 2025-02-26:
I would suggest putting some effort in acknowledging the remedies @AntonioConsilvio reported above by extending the [documentation](https://icub-tech-iit.github.io/documentation/icub_firmware/firmwareupdater/firmwareupdater/).

Issues are nice for tracking down problems, but then only experienced users tend to sift them out in search of something relevant when troubleshooting.

cc @marcoaccame

pattacini, 2025-03-02:
> I would suggest putting some effort in acknowledging the remedies [@AntonioConsilvio](https://github.com/AntonioConsilvio) reported above by extending the [documentation](https://icub-tech-iit.github.io/documentation/icub_firmware/firmwareupdater/firmwareupdater/).

Done in https://github.com/icub-tech-iit/documentation/pull/396.

AntonioConsilvio, 2025-03-05:
I think that this issue can be closed! ✅

closed 2025-03-05 by AntonioConsilvio