robotology/mesh-support #2044: [Left thumb boken]

valegagge, 2025-02-27:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Yesterday, in the calibration phase the left arm didn't calibrate and the leg bumped into the thumb and broke it.

### Detailed context

 Sorry, but I haven't an image. :(

edersonferrante, 2025-02-27:
The pin that holds the finger has broken, as seen in the CAD model it is a single piece so the entire part ( Cod. 18216 Alias  IC_028_P_147) at the base of the finger needs to be replaced.

[image]

We will update you soon on when it will be possible to replace it. 
@Fabrizio69

Gandoo, 2025-02-28:
ciao, at the moment to allow @isorrentino  to test, to  were disconnected fingertip, Fap and tendon to remove the complete finger.

[image]

cc @AntonioConsilvio @DanielePucci @Fabrizio69 @maggia80 @S-Dafarra

Gandoo, 2025-03-06:
ciao, the forearm sx was reassembled on the robot and turned on by @isorrentino to check the connections.
**I remind everyone that @AntonioConsilvio has not yet done the joint calibration.** 

[image]

[image]

cc @S-Dafarra @fbiggi @DanielePucci @maggia80 @GiulioRomualdi @Fabrizio69

AntonioConsilvio, 2025-03-19:
Joint calibrated and tested. Closing!

closed 2025-03-19 by AntonioConsilvio