robotology/mesh-support #2046: Neck broken

LoreMoretti, 2025-02-28:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Neck tendon broken. 

[image]

### Detailed context

It likely happened after a fall during walking.

### How does it affect you?

We will exclude the neck from calibration.

Gandoo, 2025-02-28:
ciao at the moment with @LoreMoretti   we have mounted the neck lock, will follow repair work

[image]

cc @isorrentino @AntonioConsilvio @DanielePucci

edersonferrante, 2025-03-04:
- [x] Neck intervention successfully completed.

[image]