robotology/mesh-support #2049: Right ankle pin got misaligned

Giulero, 2025-03-04:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The pin on the back of the right ankle pitch is misaligned, so the joint cannot calibrate properly. We are using the one at the back since the one on the front is broken (#2045).

[image]

### Detailed context

.

closed 2025-03-06 by edersonferrante

edersonferrante, 2025-03-06:
We fixed the misalignment of the pin by replacing the screw ( Cod. 17189 Alias M3X40-P-ISO4762-12_9-BLACK), as the old one had deformed.

@Fabrizio69

AntonioConsilvio, 2025-03-10:
Hi, here a pictures that shows the deformed screw:

[image]

cc @Fabrizio69 @fiorisi