robotology/mesh-support #2052: Lidar does not work / transmit data

vigisushrutha23, 2025-03-06:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

No data is received from the lidar. Two full robot/ head PC restarts have been performed but the issue persists.

### Detailed context

The error as seen in the screenshot is observed on running 'yarprobotinterface --config sensors.xml' on ergocub-head.  

[image]

### How does it affect you?

Without a functional lidar, navigation tests will not be possible.

SimoneMic, 2025-03-19:
Now it's working properly on SN002. After the replacement of the Orin it fixed it, I think.

AntonioConsilvio, 2025-03-21:
Hi! Great, so we can close this issue.

> Same error was observed on ErgocubSN001 as well.

If the problem is present also on ergoCubSN001, please open another issue for that robot if the problem is blocking for the experiments.

closed 2025-03-21 by AntonioConsilvio