robotology/mesh-support #2053: Left thumb adduction encoder reads constantly zero

S-Dafarra, 2025-03-07:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

The left thumb adduction encoder seems either broken or disconnected. It reads constantly zero

### Detailed context

The joint is controllable, but goes instantly to the limit because the encoder is constantly zero

Gandoo, 2025-03-12:
ciao @AntonioConsilvio , hello, I changed the cable between FAP and MMA which is to be replaced has the broken connector

[image]

- the FAP TUMB does not have a centering pin to hold it in place

[image]

cc  @S-Dafarra @Fabrizio69 @maggia80 @DanielePucci

AntonioConsilvio, 2025-03-18:
Here the commit that change the connection because we had problems with the MMA connector:

- https://github.com/icub-tech-iit/robots-configuration/commit/a5c2981ba242260e27d53693a275dd002734e19d

Now the FAP is working properly.

closed 2025-03-19 by AntonioConsilvio