robotology/mesh-support #2054: Torso Yaw Joint Motor Mechanics Broken

isorrentino, 2025-03-07:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The torso yaw joint was rotating only in one direction and was unable to move back.

[image]

We disconnected the motor phases to be sure that the problem was not there. After that we unmounted the torso cover and we found the broken piece in the image.

[image]

[image]

### Detailed context

The joint is now completely blocked in that position.

Gandoo, 2025-03-11:
**attention the joint is without mechanical clash**

[image]

cc @isorrentino @S-Dafarra @Fabrizio69 @maggia80 @DanielePucci @AntonioConsilvio @GiulioRomualdi

Gandoo, 2025-03-11:
detached forearm for maintenance  

[image]

cc @isorrentino @S-Dafarra @Fabrizio69 @maggia80 @DanielePucci @AntonioConsilvio @GiulioRomualdi

Gandoo, 2025-03-12:
Ciao, new power cable has been assembled and tested

[image]

cc @isorrentino @S-Dafarra @Fabrizio69 @maggia80 @DanielePucci @AntonioConsilvio @GiulioRomualdi

fbiggi, 2025-04-24:
Mechanical stop arrived from [MWS](https://github.com/icub-tech-iit/tickets/issues/4021)
Monday 04/28/2024 I booked ergoCub 1.0 S/N:001 to be able to reset the   torso jaw joint

@Fabrizio69 @maggia80 @S-Dafarra

fbiggi, 2025-04-28:
-  Processing table

- [x] Uncovering

- [x] Robot sectioning

- [x] Electrical wiring

- [x] Hardstop replacement

- [x] Upperbody lowerbody with reassembly

- [ ] Electrical wiring

- [ ] Covering

[image]

[image]

[image]

[image]

[image]

Gandoo, 2025-04-28:
ciao, To dissect the robot, the waist and X_Sens boards were disconnected and then reconnected as originally.

Robot ready for testing

[image]

[image]

[image]

@fbiggi  @Fabrizio69 @maggia80 @S-Dafarra @AntonioConsilvio

filippoborgogni, 2025-05-06:
Hi @isorrentino ,
today I tested the robot and it works properly.
Closing!🚀

cc @AntonioConsilvio