robotology/mesh-support #2056: [Robot does not start because the 2focs of board 10.0.1.6 are not visible]

fils99, 2025-03-11:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

As per title.
cc @GiulioRomualdi 

### Detailed context

```console

[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=48s 882m 663u : CFG: CANdiscovery cannot find  4 missing eobrd_foc boards for 536 ms in CAN1:1 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:12 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:23 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:34 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:4
[ERROR] embObjMultipleFTsensors: BOARD right_leg-eb6-j0_3 (IP 10.0.1.6)   open() fails to serviceVerifyActivate... cannot continue 
[ERROR] |yarp.dev.PolyDriver|right_leg-eb6-j0_3-strain| Driver <embObjMultipleFTsensors> was found but could not open
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=48s 882m 800u : CFG: EOtheMotionController cannot be configured. CANdiscovery of 2foc boards fails. see CANdiscovery messages for more details
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=48s 882m 935u : CFG: EOtheMotionController cannot be configured. CANdiscovery of 2foc boards fails. see CANdiscovery messages for more details
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=48s 883m 51u : CFG: CANdiscovery cannot find  4 missing eobrd_foc boards for 536 ms in CAN1:1 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:12 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:23 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:34 of 4: missing eobrd_foc BOARD 10.0.1.6:CAN1:4

```

### Additional context

[log.txt](https://github.com/user-attachments/files/19190530/log.txt)

Gandoo, 2025-03-12:
ciao, @AntonioConsilvio Removed right leg cover, detached black can wire to the board

[image]

Rewired the wire, replaced in connector and connected to the board. 

[image]

cc @S-Dafarra @Fabrizio69 @DanielePucci

fils99, 2025-03-12:
Thanks @Gandoo , the robot now seems working fine!

Fabrizio69, 2025-03-13:
Thanks @fils99 for your feedback, I'm closing this issue

closed 2025-03-13 by Fabrizio69