robotology/mesh-support #2057: Ethernet communication errors between robot torso and ems boards

isorrentino, 2025-03-12:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

We noticed several prints of ethernet communication errors while using the robot. Yesterday all the boards after the right arm disappeared from FirmwareUpdater and @AntonioConsilvio noticed that there is a problem on one ethernet of the right arm.

### Detailed context

here the log 

[errors_network_yri.txt](https://github.com/user-attachments/files/19209736/errors_network_yri.txt)

cc @S-Dafarra @GiulioRomualdi @traversaro @AntonioConsilvio

isorrentino, 2025-03-13:
Today we saw the error due to this issue several times. Here the screenshot of the errors on the interface. When we have this error, we are not able to correctly run our applications.

[image]

LoreMoretti, 2025-03-19:
From FirmwareUpdater we can only see the following boards:

[image]

cc @AntonioConsilvio @Gandoo @fils99

LoreMoretti, 2025-03-19:
Mistery solved. @AntonioConsilvio is working on the ethernet connections

AntonioConsilvio, 2025-03-19:
HI! We fixed the cable E3 and E22. The robot seems working properly now! Closing! ✅

closed 2025-03-19 by AntonioConsilvio