robotology/mesh-support #2058: Broken r_ankle_pitch pin

fils99, 2025-03-12:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

While controlling the r_ankle_pitch joint via current control, the pin got broken.

cc @GiulioRomualdi 

### Detailed context

[image]

LoreMoretti, 2025-04-14:
We disabled AMO for calibration: https://github.com/icub-tech-iit/robots-configuration/commit/f45a64233d1c2028857c7a599fe4d2b35f5d0259

edersonferrante, 2025-04-15:
Hi @fils99 , following the steps of the issue [2045](https://github.com/robotology/icub-tech-support/issues/2045) I removed the split screw #17189, then put back the pin #10518 and a new screw + threadlocker and tightened it with the torque wrench to 2Nm.

 [image]

[image]

@Fabrizio69

AntonioConsilvio, 2025-04-15:
Good job! 🚀