robotology/mesh-support #2059: Neck joint

ddetommaso, 2025-03-13:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

During maintenance yesterday, an issue was found in the neck.

### Detailed context

Manually moving the neck while off, the yaw seems loose.

### How does it affect you?

Currently the issue seems not affecting our activities but we will need to show a demo in the next following days. We will test the robot further in these days.

AntonioConsilvio, 2025-03-13:
When the motors are not controlled, the backlash is noticeable, see video:

https://github.com/user-attachments/assets/e1d67728-7948-462c-933f-085362be928a

However, for now the neck is usable, see video:

https://github.com/user-attachments/assets/b8b9653b-7335-45d4-baa4-0bafa71320a5

Agreed with @ddetommaso, we will leave the robot like this until it becomes a blocking issue.

ddetommaso, 2025-03-24:
@AntonioConsilvio we are experiencing issues with the neck when the robot is on. Do you need any further test to make to understand better the situation? We would need to have the robot fully working this week because on the 28th we have an important demo. Could you please schedule an intervention in Erzelli?

AntonioConsilvio, 2025-03-24:
Hi @ddetommaso, We have a pretty clear idea of what might have happened to the joint, so if we were to do an intervention we would already know where to start. 
We will have to open the whole head to reach the motor, but the only availability we have is Thursday 27th in the morning. 
After lunch we will be at Erzelli, but busy with another activity, so we will most likely not be able to provide support in the afternoon.

Without unexpected events, the intervention is possible to be finished in the morning, but if there are unforeseen circumstances, unfortunately we won't be able to do a lot in the afternoon.

Let us know if this is OK with you!

ddetommaso, 2025-03-25:
Hi @AntonioConsilvio 

thank you for your availability.
Given the importance of the demo I would please ask you guys to come on Thu for fixing the issue.
Please let me know if I can do something in the meantime to make your intervention more efficient.

AntonioConsilvio, 2025-03-25:
Great! See you on Thursday!

AntonioConsilvio, 2025-03-27:
Hi! Me and @edersonferrante fixed this problem! Now the backlash has significantly decreased, it is now possible to perform the demo without any neck movement problems!

https://github.com/user-attachments/assets/3cd34bbb-f70b-4178-b92f-e1a1f7155410

https://github.com/user-attachments/assets/2dac722d-443f-483e-b4a0-2360087f2af7

Closing! ✅

closed 2025-03-27 by AntonioConsilvio