robotology/mesh-support #2060: ergocub-torso cpu sometimes seem not working properly

fils99, 2025-03-13:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

ergocub-torso cpu sometimes seem not working properly

### Detailed context

Sometimes the ergocub-laptop disconnects from ergocub-torso cpu (connection via ethernet). Moreover, the cooling fan seems too noisy.

fils99, 2025-03-13:
Additionally, the torso computer appears as follows:

[image]

This is probably due to a hit the robot received during an experiment in current control.

AntonioConsilvio, 2025-03-19:
Hi! The connector was quite detached. Now is well connected. Closing! ✅

[image]

closed 2025-03-19 by AntonioConsilvio