robotology/mesh-support #2061: Neck movement issue

Ep3896, 2025-03-13:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

It was found that there is an issue on the movement of the neck during demo sessions. 

### Detailed context

cc @ddetommaso 

### Additional context

Unfortunately, I am not able to provide further details for this issue but the technicians/assistants present today can add more information if needed.

### How does it affect you?

It is an issue better to solve

AntonioConsilvio, 2025-03-13:
Hi @Ep3896, if you try to move the neck by hand with the robot switched off, does the neck mechanism move or does it get stuck?

Ep3896, 2025-03-13:
@AntonioConsilvio The assistant in Boggiano Pico tried to move the neck by hand with the robot switched off and it moves up-down and left-right without any constraint

ddetommaso, 2025-03-14:
The robot has recently presented this anomaly although there have been no changes to the protocol. You hear a continuous noise coming from the joints of the neck.

closed 2025-03-25 by edersonferrante