robotology/mesh-support #2065: Both wrists partially miscalibrated, maybe worn down

S-Dafarra, 2025-03-13:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

Both wrists appear miscalibrated when in home position.

[image]

[image]

### Detailed context

It seems the same issue of #2063, probably with the usual belt worn down.

cc @fiorisi @maggia80

S-Dafarra, 2025-03-13:
I just rechecked and they seem ok. Maybe it was just a weird configuration

S-Dafarra, 2025-03-13:
Nonetheless, from time to time, the right wrist moves by itself

AntonioConsilvio, 2025-03-21:
Hi! We checked the sensor wiring. A few connectors were loose, but nothing damaged. We tested the robot for a while and the wrist did not move on its own after this check.

Please give us feedback if this happens again.

AntonioConsilvio, 2025-03-24:
Throughout Friday's demo, the problem did not seem to occur. 

If @S-Dafarra experienced the same thing, I would proceed with closing the issue and then reopen it if the problem occurs again.

S-Dafarra, 2025-06-17:
Today I think it finally broke down during a demo, initially it appeared not configured, and then it went into hardware fault.

AntonioConsilvio, 2025-07-09:
Hi @S-Dafarra, have you noticed if this problem has recurred, or have we solved it?

S-Dafarra, 2025-07-09:
I have used this robot a bit less lately. I think I might have "heard" it moving once, but I am not sure. We can close this and reopen it if it happens again!

AntonioConsilvio, 2025-07-09:
Thank you! I am closing this issue, but feel free to reopen it if it happens again.

closed 2025-07-09 by AntonioConsilvio