robotology/mesh-support #2069: Oscillation of the Right Hand Fingers

Ep3896, 2025-03-13:
### Robot Name 🤖

iCubGenova08 S/N:043

### Request/Failure description

Following today's intervention, it was found that there is an issue with the right hand fingers oscillating unexpectedly. This problem causes the robot to drop the object it's grasping. 

### Detailed context

cc @ddetommaso 

### Additional context

Unfortunately, I am not able to provide further details for this issue but the technicians present today can add more information if needed.

### How does it affect you?

The issue is critical because it highly affects our current data collection

[FINGER_Issue.txt](https://github.com/user-attachments/files/19231559/FINGER_Issue.txt)

[image]

[image]

ddetommaso, 2025-03-14:
In addition to the conditions described above, the issue is recurring but not all the times,  specifically it seems due to external solicitations. Please, consider that with this robot the right hand is often problematic - let's consider a quick fix in the short terms but we will absolutely need a more in deep fix in the mid-terms.

@Antonio Could you please schedule an intervention in the next following days? 

cc @Cater19 @SilviaMoretti @AgnieszkaWykowska

ddetommaso, 2025-03-18:
@AntonioConsilvio can you please provide us an estimate for this intervention. The situation is quite limiting for us.

AntonioConsilvio, 2025-03-18:
Hi @ddetommaso, sorry for taking a few days to reply, but we've been discussing internally when to perform the intervention. 
We have some very important demos to prepare for this week, so our availability would be Tuesday or Wednesday next week.

Let us know if these days are good for you.

ddetommaso, 2025-03-18:
Hi @AntonioConsilvio,

thank you for the availability. Tuesday 25th is ok with us.
cc @gdalisca @Cater19 @SilviaMoretti @AgnieszkaWykowska

AntonioConsilvio, 2025-03-18:
Great!

gdalisca, 2025-03-24:
Hi @AntonioConsilvio! I just wanted to verify with you at what time should we meet for the intervention on the robot tomorrow. Let me know!

AntonioConsilvio, 2025-03-24:
Hello @gdalisca! We were thinking to start the intervention at 8:30 if that's ok with you.

gdalisca, 2025-03-24:
Hi! Yes, sounds perfect. I'll see you there at 8.30

AntonioConsilvio, 2025-03-25:
Hi @Ep3896 @ddetommaso @gdalisca! Me and @edersonferrante fixed this issue. We calibrate again the joint and change the control PID to avoid strange oscillation. Here the videos of the robot:

https://github.com/user-attachments/assets/34eb1d0d-ba01-4bbb-859a-36c6071bf9c2

https://github.com/user-attachments/assets/a69bd161-e693-4e57-a752-0e37333c35f1

We tested the robot several times and we didn't see oscillations anymore.

> [!Important]
> Here the PR for the configuration files:
> - https://github.com/robotology/robots-configuration/pull/721

AntonioConsilvio, 2025-03-26:
@ddetommaso PR merged, the changes have already been manually integrated into the configuration files, but remember to align robots-configuration when possible, to have clean configuration files inside the robot head. Closing! ✅

closed 2025-03-26 by AntonioConsilvio