robotology/mesh-support #2072: Temperature sensor issue on torso roll and left elbow

S-Dafarra, 2025-03-17:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

The torso roll and the left elbow go to hardware fault due to overheating at soon as the robot is turned on

### Detailed context

The error is 
```
[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2) time=309s 335m 918u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=torso_roll (NIB=0), Raw_temperature_value=-886)

```

Removing the sensor

S-Dafarra, 2025-03-17:
The same error also on right hip roll

cc @MSECode

S-Dafarra, 2025-03-17:
Possibly related: https://github.com/robotology/icub-tech-support/issues/2073

MSECode, 2025-03-17:
Yes,
after some experiments done both on a bench setup and on the robot we have realised that those errors are likely due to the fact that some joints might mount a version of the TDB with wrong resistor that make the output signal too low to be correctly sent on the i2C bus.
The affected joints should be:
- ergoCub001 torso_roll 
- ergoCub002 shoulder_yaw_r
- ergoCub002 elbow_r

S-Dafarra, 2025-03-21:
I am adding to this list also the torso pitch. It happened while preparing for a demo

```
[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2) time=1315s 893m 280u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=torso_pitch (NIB=1), Raw_temperature_value=-886)
```
 Here the full log 

[log_demo_ferrari.zip](https://github.com/user-attachments/files/19390244/log_demo_ferrari.zip)

S-Dafarra, 2025-03-21:
Just noticed another one from the same log. This time on the left hip pitch. This one was particularly dangerous as the robot was walking without protection during an important demo

```
[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=5368s 198m 265u :  MC: overheating. Temperature hardware limit exceeded. The motor has been turned off to prevent it from being damaged by overheating. (Joint=l_hip_pitch (NIB=0), Raw_temperature_value=-886)
```