robotology/mesh-support #2080: Issues on different AMOs

S-Dafarra, 2025-03-21:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

During a demo, we noticed several of these warnings:
```
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=123s 851m 885u :  HW - amo encoder status1 error: errors on reading on master and/or nonius track. (Joint= (NIB=0) Number of error in 10 seconds as internal crc errors: 0, i2c communication errors: 0, nonius track period consistency errors: 5000, eccessive input signal frequency errors: 0)
...
[WARNING] from BOARD 10.0.1.7 (right_leg-eb7-j4_5) time=124s 224m 632u :  HW - amo encoder status1 error: errors on reading on master and/or nonius track. (Joint= (NIB=0) Number of error in 10 seconds as internal crc errors: 0, i2c communication errors: 0, nonius track period consistency errors: 1366, eccessive input signal frequency errors: 0)
...
[WARNING] from BOARD 10.0.1.4 (left_arm-eb4-j2_3) time=152s 203m 498u :  HW - amo encoder status1 error: errors on reading on master and/or nonius track. (Joint= (NIB=0) Number of error in 10 seconds as internal crc errors: 0, i2c communication errors: 0, nonius track period consistency errors: 3, eccessive input signal frequency errors: 0)
```

### Detailed context

Here the full big log

[log_demo_ferrari.zip](https://github.com/user-attachments/files/19390149/log_demo_ferrari.zip)

S-Dafarra, 2025-03-21:
The problem is on the SN001 (I initially made a mistake in selecting the robot in the menu)

AntonioConsilvio, 2025-03-21:
cc @valegagge @MSECode

MSECode, 2025-03-21:
Oks, long story short, those warning have been added to help tracking the source of AMO related problem when we have weird readings on the encoder (especially if used as absolute). 
They are mainly thought to be helpful for who debugs those problems when they happen. 
If they tend to flood too much the logs we can think to increase the timer for the transmission. 
Then let me know if you think that they are informative or they need to be  improved in any way.

S-Dafarra, 2025-03-21:
They are not flooding the log, it is the first time I noticed because I was looking for errors in the log. I noticed them only on three joints, so I wondered if it was an issue

MSECode, 2025-03-21:
Oks, great.