robotology/mesh-support #2081: Ethernet issue on left arm

S-Dafarra, 2025-03-21:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

During a demo we noticed the following error on the left arm

```
ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  20 168 656
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 0 infos =  0 0 0
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  69  and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28 8 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  20 224 656
[WARNING] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=733s 89m 707u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 312[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  63  and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28 8 656
[WARNING] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=733s 101m 759u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 421[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  56  and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28 12 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=733s 102m 764u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 426[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  20 168 656
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  50  and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  20 0 656
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 0 infos =  0 0 0
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  42  and can overflow
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=733s 100m 893u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 322[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  42  and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28 20 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  20 0 656
[WARNING] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=733s 105m 758u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 421[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[DEBUG] from BOARD 10.0.1.24 (right_arm-eb24-j11_12) time=733s 660m 409u :  src LOCAL, adr 0,(code 0x04000001, par16 0x0009 par64 0x0000000000000009) -> DEBUG: tag01  SET CONTROLMODE
[WARNING] |yarp.devices.controlBoard_nws_yarp|left_arm-mc_nws_yarp| Number of streaming input messages to be read is  36  and can overflow
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 1, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 0 infos =  0 0 0
[... 164 lines cut ...]
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 0, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28 12 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 0, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  63737 32506 656
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 0, 
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 0, 
[WARNING] HostTransceiver::addSetROP__(): eo_transceiver_lasterror_tx_Get() detected: err= 5 infos =  28 12 656
[ERROR] embOBJerror(): errtype =  TRACE from EOobject = 
[... cut ...]
[DEBUG] HostTransceiver::addSetROP__(): eo_transceiver_OccasionalROP_Load() for BOARD /w IP [1, 0, ., 0, ., 1, ., 3, 0, 
[WARNING] from BOARD 10.0.1.30 (right_arm-eb30-j4_6) time=733s 211m 866u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 434[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
```

### Detailed context

Here the full big log 

[log_demo_ferrari.zip](https://github.com/user-attachments/files/19390208/log_demo_ferrari.zip)

S-Dafarra, 2025-03-21:
I have noticed another error in the network
```
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.1 (right_arm-eb1-j0_1) has been silent for 0.0229745 sec (its timeout is 0.02 sec)
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.6 (right_leg-eb6-j0_3) has been silent for 0.0232494 sec (its timeout is 0.02 sec)
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.8 (left_leg-eb8-j0_3) has been silent for 0.023231 sec (its timeout is 0.02 sec)
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.21 (head-eb21-j2_3) has been silent for 0.0229247 sec (its timeout is 0.02 sec)
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.1 (right_arm-eb1-j0_1) has shown after being lost for 0.0284851 sec
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.6 (right_leg-eb6-j0_3) has shown after being lost for 0.028723 sec
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.8 (left_leg-eb8-j0_3) has shown after being lost for 0.0286806 sec
[DEBUG] eth::EthMonitorPresence: BOARD 10.0.1.21 (head-eb21-j2_3) has shown after being lost for 0.0283561 sec
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 62m 393u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 62m 540u : SYS: detected a runtime error
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 62m 724u :  SYS: the TX phase of the control loop has last more than wanted. TX execution time 351[usec]. Latest previous execution times[usec] (RX=N/A, DO=N/A, TX=134); (Rx=76, DO=445, ...). Num of CAN frames[CAN1,CAN2]=[1 0]
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 63m 236u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 389[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[ERROR] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 64m 635u :  SYS: the board has detected a hole in rx ropframe sequence number. Expected number is 7930548, received number is 694 
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 65m 124u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 512[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[WARNING] from BOARD 10.0.1.8 (left_leg-eb8-j0_3) time=8250s 66m 980u :  SYS: the RX phase of the control loop has last more than wanted. RX execution time 367[usec]. Latest previous execution times[usec] (..., Tx=0);(RX=0, DO=0, TX=0);
[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 3, .,  , E, x, p, e, c, t, e, d, :,  , 8, 1, 8, 2, 0, 2, 1, ,,  , R, e, c, e, i, v, e, d, :,  , 8, 1, 8, 2, 0, 2, 6, ,,  , M, i, s, s, i, n, g, :,  , 5, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 2, 5, 2, 2, 7,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 3, 1, 2, 2, 7,  , u, s, 
[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 4, .,  , E, x, p, e, c, t, e, d, :,  , 8, 1, 7, 9, 4, 1, 9, ,,  , R, e, c, e, i, v, e, d, :,  , 8, 1, 7, 9, 4, 2, 5, ,,  , M, i, s, s, i, n, g, :,  , 6, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 1, 2, 8, 9, 6, 5,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 1, 3, 5, 9, 6, 5,  , u, s, 
[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 5, .,  , E, x, p, e, c, t, e, d, :,  , 8, 1, 8, 1, 7, 6, 5, ,,  , R, e, c, e, i, v, e, d, :,  , 8, 1, 8, 1, 7, 7, 1, ,,  , M, i, s, s, i, n, g, :,  , 6, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 6, 5, 2, 1, 8,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 7, 2, 2, 1, 8,  , u, s, 
[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 2, .,  , E, x, p, e, c, t, e, d, :,  , 8, 1, 8, 2, 1, 7, 6, ,,  , R, e, c, e, i, v, e, d, :,  , 8, 1, 8, 2, 1, 8, 1, ,,  , M, i, s, s, i, n, g, :,  , 5, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 3, 6, 4, 5, 7,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 4, 2, 4, 5, 7,  , u, s, 
[ERROR] [h, o, s, t, T, r, a, n, s, c, e, i, v, e, r, (, ), :, :, p, a, r, s, e, (, ),  , d, e, t, e, c, t, e, d,  , a, n,  , E, R, R, O, R,  , i, n,  , s, e, q, u, e, n, c, e,  , n, u, m, b, e, r,  , f, r, o, m,  , I, P,  , =,  , 1, 0, ., 0, ., 1, ., 2, 4, .,  , E, x, p, e, c, t, e, d, :,  , 8, 1, 8, 1, 8, 6, 9, ,,  , R, e, c, e, i, v, e, d, :,  , 8, 1, 8, 1, 8, 7, 5, ,,  , M, i, s, s, i, n, g, :,  , 6, ,,  , P, r, e, v,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 2, 8, 3, 5, 1,  , u, s, ,,  , T, h, i, s,  , F, r, a, m, e,  , T, X,  , a, t,  , 8, 2, 5, 0, 3, 3, 5, 3, 5, 1,  , u, s, 
[... cut ...]
[ERROR] theNVmanager::Impl::ask() had a timeout for BOARD right_arm-eb3-j2_3 IP 10.0.1.3 and nv ID32 = 0x0100000b -> IND = 0, TAG = eoprot_tag_mc_joint_status_target
[WARNING] EthResource::getRemoteValue() cannot have a reply from BOARD right_arm-eb3-j2_3 with IP 10.0.1.3 at attempt # 1 w/ timeout of 0.2 seconds
[ERROR]   FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD right_arm-eb3-j2_3 with IP 10.0.1.3  even after 0.200191 seconds: CANNOT PROCEED ANY FURTHER
[ERROR] embObjMotionControl::getRefVelocityRaw() could not read reference vel for  BOARD right_arm-eb3-j2_3 (IP 10.0.1.3)  joint  0
```

AntonioConsilvio, 2025-07-16:
Hello! The robot has performed several demos in the last period without presenting the problem.

We probably fixed the ethernet problems during another maintenance intervention. Closing!