robotology/mesh-support #2082: r_ring_pinky_oc goes to overcurrent

S-Dafarra, 2025-03-21:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Here the error:

```
[ERROR] from BOARD 10.0.1.22 (right_arm-eb22-j7_10) time=5199s 413m 348u :  MC: overcurrent. The motor has been turned off to prevent it from being damaged by an impulsive spike of current. (Joint=r_ring_pinky_oc (NIB=3))
```

### Detailed context

Here the big full log 

[log_demo_ferrari.zip](https://github.com/user-attachments/files/19390296/log_demo_ferrari.zip)

PasMarra, 2025-04-01:
https://github.com/robotology/icub-tech-support/issues/2037#issue-2874607777

filippoborgogni, 2025-04-02:
Hi @S-Dafarra ,
I worked to solve this issue and [this one](https://github.com/robotology/icub-tech-support/issues/2037) and after various tests also using the jogger board I could observe that the problem does not seem to be hardware-related, because using the JOGGER board I could observe that the finger moves correctly. 

cc @AntonioConsilvio @Fabrizio69

S-Dafarra, 2025-04-03:
> problem does not seem to be hardware-related

I think we cannot exclude this yet. I believe that the jogger does not care too much about the current used to move the motor, while the current readings on the motorgui are too seldom to actually check if we are reaching the limit.

Maybe there is some increased friction in the mechanism?

closed 2025-05-30 by filippoborgogni