robotology/mesh-support #2083: Left shoulder yaw joint limit pin broke

HosameldinMohamed, 2025-03-24:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The socket in which the pin acts as the hard stop for the left shoulder yaw has fractured

### Detailed context

[image]

[image]

### Additional context

Here's the piece in the CAD

[image]

Here's the name and code in Wingst

[image]

### How does it affect you?

Unable to perform joint calibration type 10, which blocks iRonCub experiment

CC @pillai-s

fbiggi, 2025-03-25:
On Wednesday 26/03/2025 we intervene on the robot to replace the code 12279

@Fabrizio69 @maggia80

Gandoo, 2025-03-26:
ciao, Disconnected the wiring in order to replace the mechanical block, below reconnected.

[image]

fbiggi, 2025-03-26:
Below are the various processing phases:
- [x] Unwiring of the left arm @Gandoo 
- [x] Dismantling of the hard stop on the bench
- [x] Replacing the compliant hard stop
- [x] Mechanical reassembly on the robot + wiring @Gandoo 

[image]
[image]
[image]
[image]
[image]

@Fabrizio69 @maggia80

AntonioConsilvio, 2025-03-26:
Hi @HosameldinMohamed, please give us, feedback if everything works as expected after the intervention!

HosameldinMohamed, 2025-03-26:
Ciao @AntonioConsilvio @fbiggi @Gandoo  all is good 🚀
We started the robot and the calibration type 10 works as expected 🎉

Many thanks!

AntonioConsilvio, 2025-03-26:
Great! Good job @Gandoo and @fbiggi 🚀

closed 2025-03-26 by AntonioConsilvio