robotology/mesh-support #2087: Right arm shoulder roll issue

elandini84, 2025-03-26:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

For some reason the right shoulder roll joints of the robot seems to "shift" it's lower limit bringing the arm closer to the torso.

### Detailed context

It appears that the right shoulder roll joints randomly changes its lower limits. This results in the arm going closer to the torso and, when this happens during scripted movements, in unforseen collisions between the forearm/hand and the torso. This has already happend at least 3 times but we weren't able to determine the exact conditions that triggered this strange behavior.

### How does it affect you?

This issue can lead to issues during tests and demos and can easily damage the robot. We have an event from the 10 to the 12 of April in Genova and we would need this problem addressed by then in order to be as safe as possible during the demos with the guests.

ste93, 2025-04-08:
hi @elandini84, on friday it happened that the roll suddenly went to zero while the encoder was still at 50, it seems a problem related to the wrong encoder value as https://github.com/robotology/icub-tech-support/issues/2104

AntonioConsilvio, 2025-07-10:
Hi, We worked on this issue some time ago checking the wiring, do you know if the problem has recurred recently?

AntonioConsilvio, 2025-07-16:
I proceed to close the issue as the problem was no longer reported. 
If it occurs again, please feel free to reopen the issue. Closing!

ste93, 2025-07-18:
Hi @AntonioConsilvio, sorry for the delay. I haven't worked on the robot that much but no one has reported any issue on this, so in case we will re-open it. Thank you